Ultimaker/Cura #10363: can't start cura, Python error in main script

mofada, 2021-08-30:
### Application Version

4.11.0/4.10

### Platform

windows 10

### Printer

Anycubic

### Reproduction steps

can't start cura, Python error in main script

### Actual results

can‘t run

### Expected results

run cura

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[image]

mofada, 2021-08-30:
My computer does not have a user named `ultimaker`, my user name is `fada`

fvrmr, 2021-08-30:
Hi @mofada could you try to delete Cura `C:\Program Files\Ultimaker Cura 4.11.0` and do and clean install? 
Did you have this issue with previous versions of Cura?

mofada, 2021-08-30:
I installed it in `D:\Program Files\Ultimaker Cura`, I used `4.9` before, and when I saw a new version, I tried to update it, and then this error occurred, I need to uninstall and empty the folder Then reinstall it?

> Hi @mofada could you try to delete Cura `C:\Program Files\Ultimaker Cura 4.11.0` and do and clean install?
> Did you have this issue with previous versions of Cura?

mofada, 2021-08-31:
Reinstallation can solve the problem

closed 2021-08-31 by mofada

Ghostkeeper, 2021-09-08:
> My computer does not have a user named `ultimaker`, my user name is `fada`

The "ultimaker" user comes from the file path of `cura_app.py` which got packed into the cura.exe application and doesn't actually exist any more. It was the path on Ultimaker's build server before it got packaged inside.

Installing Cura on top of an existing installation, in the same folder as a previous version of Cura, is not a good idea. It'll override the files that have changed, but if any files had been removed in the new version they would still be there in your installation, unmodified by the installation of a newer version. That may have been why this went wrong.