Ultimaker/Cura #10364: Incredibly laggy performance on Mac, log shows infinite loop of "Socket was unable to bind to port"

melyux, 2021-08-30:
### Application Version

4.10.0

### Platform

Mac OS X 10.15.7

### Printer

Ender 3

### Reproduction steps

I tried to start Cura. I deleted the configuration and tried it fresh. No difference.

### Actual results

Cura takes about 60 seconds to launch, and uses 100% CPU and takes 2.2 GB of RAM, with no model or anything loaded. The whole interface is unresponsive and laggy, and clicks take several seconds to register. Everything is unusably slow.

The `cura.log` shows a constant infinite loop of

```
2021-08-30 03:02:05,648 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [341]: Attempting to kill the engine process
2021-08-30 03:02:05,652 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-08-30 03:02:05,658 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [216]: Socket was unable to bind to port, increasing port number to 50218
2021-08-30 03:02:05,661 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-08-30 03:02:05,663 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [341]: Attempting to kill the engine process
2021-08-30 03:02:05,666 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-08-30 03:02:05,671 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [216]: Socket was unable to bind to port, increasing port number to 50219
2021-08-30 03:02:05,673 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-08-30 03:02:05,675 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [341]: Attempting to kill the engine process
2021-08-30 03:02:05,678 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-08-30 03:02:05,682 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [216]: Socket was unable to bind to port, increasing port number to 50220
2021-08-30 03:02:05,685 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
```

forever while Cura is on. The attempted port number keeps increasing.

### Expected results

Cura should be responsive and not use 100% CPU.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Included above

fvrmr, 2021-08-30:
Hi @melyux could you also check if this happens in 4.11 beta? 
You can download the beta here: https://github.com/Ultimaker/Cura/releases/tag/4.11-beta

melyux, 2021-08-30:
Hey @fvrmr, I tried the 4.11 beta with no change. Log's still full of it and the interface is really slow.

nallath, 2021-08-31:
Do you have an active VPN, firewall or virus scanner?

melyux, 2021-08-31:
I don't have one running at the same time as Cura, nope.

nallath, 2021-08-31:
Huh. That's odd. I don't get why it's then unable to create a socket to the backend. In all cases that previously happened, it was either of those three that caused it.

melyux, 2021-08-31:
It was working a few months ago when I last tried. It's working on a second Mac too. Anything I can do to help debug?

nallath, 2021-09-01:
Could you include the entire log file? Maybe there is something there that indicates what started the loop.

melyux, 2021-09-01:
Sure, it's attached because of its sheer size, but it really doesn't have much other than the repeated 4 lines, continuing until you quit the app.

[cura-1.log](https://github.com/Ultimaker/Cura/files/7094925/cura-1.log)

nallath, 2021-09-02:
Could you delete the log files, try to start Cura and quickly kill it? I'm looking for something in the logs that happens before this loop spam.

melyux, 2021-09-02:
Done. Looks like it only starts logging after the FileLogger plugin is loaded, so it probably doesn't show anything before that:

```
2021-09-02 12:57:06,041 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-09-02 12:57:06,044 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [216]: Socket was unable to bind to port, increasing port number to 49720
2021-09-02 12:57:06,047 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-09-02 12:57:06,048 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [354]: Attempting to kill the engine process
2021-09-02 12:57:06,051 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-09-02 12:57:06,057 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [216]: Socket was unable to bind to port, increasing port number to 49721
2021-09-02 12:57:06,060 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-09-02 12:57:06,062 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [354]: Attempting to kill the engine process
2021-09-02 12:57:06,065 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-09-02 12:57:06,069 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [216]: Socket was unable to bind to port, increasing port number to 49722
2021-09-02 12:57:06,071 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-09-02 12:57:06,072 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [354]: Attempting to kill the engine process
2021-09-02 12:57:06,075 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-09-02 12:57:06,080 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [216]: Socket was unable to bind to port, increasing port number to 49723
2021-09-02 12:57:06,083 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-09-02 12:57:06,084 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [354]: Attempting to kill the engine process
2021-09-02 12:57:06,087 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
```

[cura.log](https://github.com/Ultimaker/Cura/files/7101447/cura.log)

nallath, 2021-09-03:
That is a lot less information than I was hoping for. I'm out of ideas of what it could be though. It almost has to be something that is different on your machine, as it is working for other people. Are you very sure you don't have some strange routing / VPN / network / security stuff setup?

melyux, 2021-09-03:
I don't have anything like that running. Is there something else I can run to diagnose what's preventing the socket connections?

nallath, 2021-09-03:
I'm not super experienced (read; I know almost nothing about it) with OSX. So I don't have any tips for that. Maybe someone else has an idea how to figure that out.

melyux, 2021-09-03:
It's working! I tried everything, and finally installed a security update for the OS and it's working now. I don't really understand, but that's good.

Ghostkeeper, 2021-09-08:
Cura uses a local socket connection to communicate between its front-end and its slicing engine. It sends a 3D model to the engine, and at some point receives g-code back. To do that, CuraEngine listens to port 127.0.0.1:49674 and Cura's front-end connects to that too.

Some security software sees this as an internet-facing operation, and then proceeds to block it. Since it's 127.0.0.1 (localhost), this connection does not go over the internet but stays inside your computer. It doesn't accept connections from the outside. It is just as safe as communication via stdout, and safer than communication via hard disk. This socket connection can be trusted as long as the running applications on your computer can be trusted.

My guess is that a previous MacOS update had a bug there that would also block this local connection, but this bug got fixed in your new OS update.
It's good to know that this has been an issue on MacOS and the solution would be to make sure security updates for MacOS are up-to-date.

closed 2021-09-08 by Ghostkeeper