Ultimaker/Cura #10369: Extra top layer in the middle of model

pemurr, 2021-08-31:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creality-3 Pro

### Reproduction steps

Load thingiverse model 1743145, specifically HIVE_module_V3.stl (older version than current one).
Scale up to 101%
Using Standard Quality profile, slice.

Rescale to different magnitudes and test by slicing

Use a different quality profile and test by slicing

Return to Standard Quality, switch to 0.21 layer height and reslice

### Actual results

See an extra unexpected top layer being added in the middle of the model

Using 101% scaling, this is added at layer 58 of 389.

At 97% scale, the extra top layer appears at layer 62 of 373.  Other scales may or may not introduce the extra top layer, but at widely varying points.

Using a higher or lower quality profile (different layer heights), the effect is not seen

Using Standard Quality, but modifying layer height to 0.21mm, the effect disappears at all scales tested.

Failed project file attached, plus preview

[Hive_fail.zip](https://github.com/Ultimaker/Cura/files/7083914/Hive_fail.zip)

### Expected results

The output should not have the extra top layer, ie appears as shown in success attachment (plus project file)

[Hive_success.zip](https://github.com/Ultimaker/Cura/files/7083906/Hive_success.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

If sliced at 100% scaling, the issue does not appear.

I have tried various permutations of scaling and quality - only appears at 0.2mm layer height, but the presence or absence of the extra top layer and its position in the output can vary with scale.  Switching initial layer height to 0.25mm or more also removes the effect.

fvrmr, 2021-09-02:
Hi @pemurr thank you for your report. 
I opened your project and the model from Thingiverse but v5 instead, which looks the same. And it didn't happen on the new model. 
[image]

Can you check if that works for you?

But I will bring this up for discussion with the team. Keep you posted!

pemurr, 2021-09-02:
A fresh download of the V5 model produces the extra skin here at layer 58 for a 101% scaling and 0.2mm quality (100% is fine, and 0.21mm layer height is also OK).  At 99% scale it happens at layer 294 this time.  Tested a few scales around the mid point:

95% multiple layers - 150, 242, 265, 298
96% layer 95
97% no error (different to my previous V3 result)
98% layer 95
99% layer 294
100% no error
101% layer 58
102, 103, 104% no error
105% layer 150

It's weird, especially that 95% scaling.  I've dropped the project file for that one here.  

I didn't see any problems at any scale as soon as I switched the layer height to 0.21 mm.

Other details that might be relevant - using Cura 4.10.0, on Alienware Windows 10 laptop, latest updates, 16 GB RAM, plenty of storage.  OK, I run a lot of stuff (IT security work), but it's mostly well out of the way, and relates to networks rather than local processing.  Any known incompatibilities?

[CE3PRO_HIVE_Module_V5_95scale.zip](https://github.com/Ultimaker/Cura/files/7099612/CE3PRO_HIVE_Module_V5_95scale.zip)

pemurr, 2021-09-04:
Even weirder.  Moving the V5 101% scaled model around on the build plane, making sure that Z remains 0, I can see versions with the extra skin at different layers, and sometimes even, if you hit the right spot (lower left quadrant seems to work best, but not always and not exclusively), no extra skin at all.  That's all, just change the X and Y coordinates and the slicing behaviour changes.  So far, without amending anything else I have seen the original layer 58 (centered), then 98 (located at back left, negative X and Y), 127 (back right, positive X, negative Y) and 144 (front right, X and Y both positive).

I did a multiply on the original to end up with five identical items on the build plane simultaneously, and while two had no extra skins the other three had superfluous content (in one case multiple extras) all at differing layers.

Do I have to tram the bed for Cura as well as the printer<g>?

[image]

pemurr, 2021-09-04:
OK, this is about as definitive as I can get without some clue as to what I should be looking at.  Take the V5 stl from Thingiverse, and scale it to 101%, then replicate it to produce five items, all without skirts or brims to fit without interference.  One placed at X,Y = 0,0, the remaining four at +/- 65 for X and Y, gives me five items that all behave differently when sliced together.  The X,Y at 65, -65 has multiple extra skins.  Only X,Y=-65,65 is correct.  Project file and image attached.

In this case, switching the layer height to 0.21mm did NOT entirely remove the effect - three out of five had extra skins.  Even going to low quality (0.28mm) left me with a faulty central item, though the other four look OK.  Using the CHEP 0.12 setting as the other extreme meant three bad outputs again.

Checking the original [Thingiverse ](https://www.thingiverse.com/thing:1743145/comments) page, someone noted the problem back in January of this year, but it doesn't appear to have been followed up.

[Hive_V5x5.zip](https://github.com/Ultimaker/Cura/files/7110036/Hive_V5x5.zip)

Ghostkeeper, 2021-09-09:
The original project file already reproduces the problem 100%. In fact, I see two bugs: It's missing the inner outline on layer 62 (the reported bug) but it's also missing the entire layer 89.

This sort of issue is often a problem with the stitching process. Basically, Cura will create cross sections of all triangles in the model, resulting in a bunch of line segments. These line segments are then combined together to form polygons. This combining is sensitive to mistakes in the model (what happens if line segments cross, etc) and also has a lot of edge cases. We could be hitting an edge case here that's not being dealt with correctly. That would explain why scaling it a tiny amount creates a different result. If a triangle has two vertices exactly on the plane of a layer, will the line segment between them be included in that layer? That's the sort of thing we should be looking for.

The problem could also be in the mesh. Tooling helps. CloudCompare is useful to check for normals being off, but that looks fine on this model. I use Blender to look at the wireframe. Layer 62 gets sliced at Z=12.3mm, which can be visualised in Blender with a plane. But I don't really see any oddities there, like intersecting lines or horizontal edges of triangles (the vague line in the screenshot here is by Blender rendering the intersection of the plane, not part of the wireframe of the model).
[image]
MeshLab is also not finding any significant issues, although MeshLab did crash on some filters on this model, but that's just MeshLab being MeshLab. So from what I can tell, the mesh is perfectly fine. And I don't see it hitting specific edge cases, but those could be invisible to the visualisations I'm using.

Next step is to look at Cura's mesh fixing features to see if any of them fix it. The mesh fixing settings didn't make a difference here. However Surface Mode set to "Surface" did generate surfaces for the missing contours, so the slicing stage did find intersections with the triangles. It must be the stitching algorithm.

To go any further I'd need to debug the intermediary stages of CuraEngine, determine in which stage of stitching the polygons are getting lost and why it discards them.

fvrmr, 2021-09-09:
We have discussed this with the team and decided to defer this, since we believed it had something to do with the model. 
@Ghostkeeper do you think those bugs are gonna change the decision we made?

pemurr, 2021-09-09:
I may be a little naïve here, but would an error in the model give such wildly different results when just changing the XY co-ordinates on the build plane for Cura?  If you create a single copy and just migrate it around the base you will see what I mean - this is perhaps a simpler demonstration of what I tried to show in my previous post.  That strongly suggests at least some inconsistency in how Cura scans things as all other parameters remain consistent, rounding errors in the calculations would be my uneducated guess.

Just to add a  little more spice, I tried the same approach with a fresh 4.11 install, and see the same behaviour - down to the same extra skins at different scalings.

smartavionics, 2021-09-09:
@pemurr, I realise that this doesn't help much but I tried slicing your project using my Cura variant (https://github.com/smartavionics/Cura/releases) on Linux and couldn't get it to fail, I moved the model around a lot and I never saw the extra layer as described above.  Could you please try with one of my builds just to confirm if it shows the same issue or not. You can install them without it affecting your UM Cura installation. Thanks.

pemurr, 2021-09-09:
@smartavionics, correct, can't get anything to break with the 'master' build.  All positions and size variations that I've tried yield an apparently clean output.  

Note that something did break (Cura just failed to slice after a few seconds, and asked me to report a bug - not replicated enough yet) when I tried the five side-by-side items, as outlined in the Hive_V5x5 post above, but that's probably a different bug - maybe memory related, I do have a lot running at the moment that I cannot just shut down.

smartavionics, 2021-09-09:
>@smartavionics, correct, can't get anything to break with the 'master' build. All positions and size variations that I've tried yield an apparently clean output.

Thanks for trying my build. So it is possible that the problem lies in some code that is different in UM's Cura and my Cura. Cura devs, I can't think of anything glaringly obvious that would cause this issue but one difference that could be relevant is that I am not using your version of simplify() as I never believed that the last set of mods you made to that method were correct. That's a long shot, for sure.

> Note that something did break (Cura just failed to slice after a few seconds, and asked me to report a bug - not replicated enough yet) when I tried the five side-by-side items, as outlined in the [...]

If it continues to occur and you can provide a cura.log or stderr.log file that contains some useful info or you have a project file that triggers the crash, please open an issue on my Cura repo. Thanks.

smartavionics, 2021-09-09:
Well, that looks to be an inspired guess. I did the obvious experiment and swapped out my simplify() for the UM version and using the project provided above with no changes we now get a couple of filled in layers...

[image]

Here's what I am using for simplify(), you may find it useful.

```
void PolygonRef::simplify(const coord_t smallest_line_segment_squared, const coord_t allowed_error_distance_squared)
{
    if (size() < 3)
    {
        clear();
        return;
    }
    if (size() == 3)
    {
        return;
    }

    ClipperLib::Path new_path;
    Point previous = path->at(0);
    Point current = path->at(1);
    /* When removing a vertex, we'll check if the delta area of the polygon
     * remains below allowed_error_distance_squared. However when removing
     * multiple consecutive vertices, each individual vertex may result in a
     * delta area below the threshold, while the total effect of removing all of
     * those vertices results in too much area being removed. So we accumulate
     * the area that is going to be removed by a streak of consecutive vertices
     * and don't allow that to exceed allowed_error_distance_squared. */
    coord_t accumulated_area_removed = previous.X * current.Y - previous.Y * current.X; //Shoelace formula for area of polygon per line segment.

    new_path.push_back(previous);

    // end points of the reference line against which other line segments are compared for co-linearity
    Point ref_line_start;
    Point ref_line_end;

    for (size_t point_idx = 1; point_idx < size(); point_idx++)
    {
        current = path->at(point_idx);

        Point next = path->at((point_idx + 1) % size());

        const coord_t length2 = vSize2(current - previous);
        if (length2 < 25)
        {
            // We're allowed to always delete segments of less than 5 micron.
            continue;
        }

        //Check if the accumulated area doesn't exceed the maximum.
        accumulated_area_removed += current.X * next.Y - current.Y * next.X; //Shoelace formula for area of polygon per line segment.

        const coord_t area_removed_so_far = accumulated_area_removed + next.X * previous.Y - next.Y * previous.X; //Close the polygon.
        const coord_t base_length_2 = vSize2(next - previous);
        if (base_length_2 == 0) //Two line segments form a line back and forth with no area.
        {
            continue; //Remove the vertex.
        }
        //We want to check if the height of the triangle formed by previous, current and next vertices is less than allowed_error_distance_squared.
        //L = b * h           [apply above two and take out the 1/2]
        //h = L / b           [divide by b]
        //h^2 = (L / b)^2     [square it]
        //h^2 = L^2 / b^2     [factor the divisor]
        const coord_t height_2 = area_removed_so_far * area_removed_so_far / base_length_2;
        if (length2 < smallest_line_segment_squared && height_2 <= allowed_error_distance_squared) //Line is small and removing it doesn't introduce too much error.
        {
            continue; //Remove the vertex.
        }
        else if (length2 >= smallest_line_segment_squared && new_path.size() > 2 &&
                LinearAlg2D::getDistFromLine(current, ref_line_start, ref_line_end) <= 2) //Almost exactly straight (barring rounding errors).
        {
            new_path.pop_back(); //Remove the previous vertex but still add the new one.
        }
        else
        {
            // update the reference line
            ref_line_start = previous;
            ref_line_end = current;
        }
        //Don't remove the vertex.

        accumulated_area_removed = current.X * next.Y - current.Y * next.X;
        previous = current; //Note that "previous" is only updated if we don't remove the vertex.
        new_path.push_back(current);
    }

    //For the last/first vertex, we didn't check the connection that closes the polygon yet. Remove it if it's too short.
    if(new_path.size() > 2 && (vSize2(new_path.back() - new_path[0]) < smallest_line_segment_squared || vSize2(new_path.back() - new_path[new_path.size() - 2]) < smallest_line_segment_squared))
    {
        if (LinearAlg2D::getDist2FromLine(new_path.back(), new_path[new_path.size() - 2], new_path[0]) < allowed_error_distance_squared)
        {
            new_path.pop_back();
        }
    }
    //For the first two points we haven't checked yet if they are almost exactly straight.
    if (new_path.size() > 2)
    {
        if (LinearAlg2D::getDistFromLine(new_path[0], new_path.back(), new_path[1]) <= 2)
        {
            // first point is colinear, remove it and check the 2nd point
            new_path.erase(new_path.begin());
            if (new_path.size() > 2 && LinearAlg2D::getDistFromLine(new_path[0], new_path.back(), new_path[1]) <= 2)
            {
                new_path.erase(new_path.begin());
            }
        }
        else if (LinearAlg2D::getDistFromLine(new_path[1], new_path[0], new_path[2]) <= 2)
        {
            new_path.erase(new_path.begin() + 1);
        }
    }

    *path = new_path;
}
```

Ghostkeeper, 2021-09-13:
I 

> We have discussed this with the team and decided to defer this, since we believed it had something to do with the model.
> @Ghostkeeper do you think those bugs are gonna change the decision we made?

I don't believe it has to do with the model, so that decision wasn't well informed. Or did you find something wrong in the model that I overlooked?

The problem with pointing out a certain algorithm (like `simplify()`) is that this case is an edge case. I could also change the slicing algorithm to always add +0.001mm to the Z coordinate of each layer, and claim that it fixes the issue because it doesn't occur any more. While in reality it doesn't occur any more because all of the vertex coordinates are then slightly different such that the edge case isn't hit any more, and it'll still occur with different settings.

It does narrow it down to something that happens at or after the `simplify()` method though. So it could be the simplify method or anything that happens to the layer afterwards. The `simplify()` method is still a good contender.

Ghostkeeper, 2022-07-18:
This issue has been fixed in Cura 5.1. The algorithm that had an integer overflow has been rewritten. I paid special attention to prevent overflows because of this error.

closed 2022-07-18 by Ghostkeeper