Ultimaker/Cura #10370: simple fix for the Gradual infill steps

dsvilko, 2021-08-31:
### Is your feature request related to a problem?

As it's currently implemented, the algorithm for placing the infill with "Gradual infill steps" turned on almost always places the new, higher density infill layer lines such that they either start, stop or are completely printed in mid-air. There is no check that there is actually something on the layer below to anchor the printed infill line. 
This makes "Gradual infill steps" basically useless for anything other than geometries with completely flat roof. This is a real pity as that option has a potential to hugely speed up the large prints that do not otherwise require dense infill for structural strength, only as internal supports for a complex roof.

### Describe the solution you'd like

I propose this simple solution: take the currently computed infill line as a starting point but extend it in both directions until it either encounters an inside wall or a previously printed infill line from the layer below. This way you insure that each infill line can not start or stop in mid-air and you still use as little material as theoretically possible.

### Describe alternatives you've considered

The "Connect infill lines" option currently helps just a small amount but the resulting often wiggly infill line shapes are still very badly supported by the previous layer or still completely impossibly to print.
Compared to the proposed working implementation of a "Gradual infill steps", "Infill support" option is extremely slow to compute, requires much more material, longer print times and is much more likely to fail as it relies on potentially thin and tall support towers that make no sense inside the object.

### Affected users and/or printers

Every time you try to find a way to print a hollow, complex object as fast as possible with as little material as possible. So, often?

dsvilko, 2021-08-31:
Closed because I found there was already a similar request, unfortunately without much interest from the developers.

closed 2021-08-31 by dsvilko

jellespijker, 2021-08-31:
Please mention that request in a comment so we can track and adjust priorities if it is requested by  more of our users

Ghostkeeper, 2021-09-10:
I'm guessing this one: https://github.com/Ultimaker/Cura/issues/9374