Ultimaker/Cura #10372: Bridging orientation issue

JAM3SY, 2021-09-01:
### Application Version

4.11

### Platform

Windows 10

### Printer

BIQU B1

### Reproduction steps

[image]
There seems to be issue where bridging is ignoring the more obvious bridge direction (shortest distance, with anchored sides) in preference for a bridge across 2 bridged skins.

I did try rotating the model before slicing at 3 different angles 15, 45, and 90 degrees. I also changed top/bottom/infill direction [0,90], however the bridge remained the same.
I also tried a wall count of 999, which resulted in a concentric bridge... again, not really desirable.

### Actual results

No change to the bridge direction.

### Expected results

I expected some change in direction of the bridge, however the parameter changed don't seem to have a bearing on bridge direction. with the exception of wall count which results in a concentric bridge.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Backplate Interface.zip](https://github.com/Ultimaker/Cura/files/7088064/Backplate.Interface.zip)

jellespijker, 2021-09-01:
Thanks for following up on this @JAM3SY, at a first glance the direction of the bridging lines are indeed not logical. My colleague @Ghostkeeper suggested that this could indicate a bug and is certainly worth looking into. Especially since you have enabled bridge settings which uses the new bridging algorithm.

fvrmr, 2021-09-09:
I have created a ticket on our backlog for this. Devs see CURA-8548

MariMakes, 2023-08-22:
Hey @JAM3SY,

Quick update from our side 👋
We are cleaning up our open issues so that we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue where the bridging doesn't show up correctly.

I was retesting your issue and noticed it doesn't show up if I use the default profiles. You changed a lot of different settings in the included project file. 
Could you help us find the setting combination that is triggering this behavior? 
With this information, we can better understand what the priority is of your issue. 
Thank you 🙏

h3nk86, 2023-08-27:
Hi @MariMakes 

I did a lot of try´n´error and found that reducing the value of _"Bridge Sparce Infill Max Density"_ almost to 0% (in my case 6%) causes a change of direction of the first Bridge Skin. Now optimal. 
However, when bringing _"Bridge Sparce Infill Max Density"_ up to 60...80% it creates a great "bridge-layer" above infill. Which results in a better surface quality, since it uses the Flow value of the bridge skin and not of "Bottom/Top". 
So that is a work around for me right now. I think everyone would appreciate a fix.
Thank you so much!

ungutknut, 2023-12-06:
Bridging only works in a few very limited scenarios and I rarely ever have a model where I can really make use of this important feature. I hope this gets some love from devs in the future.

> Could you help us find the setting combination that is triggering this behavior? With this information, we can better understand what the priority is of your issue. Thank you 🙏

This should work with 5.6.0:

Without bridging settings (standard "draft" settings for generic PLA)
[image]

Here's the same with bridging enabled (default settings except for "bridge skin support threshold" set to 90% - otherwise it won't detect the bridge):
[image]

As you can see, the lines for the bridge get oriented in the worst possible direction. From multiple attempts in making bridging work I can say that the line direction appears to be random at best. Please also take notice of the fact that the whole skin "island" is treated as a bridge - which is another major issue and again prevents me from making any use of bridging settings.

I also attached the project file for reproducing the issue.
[CFFFNOTL_extruder_testcase.zip](https://github.com/Ultimaker/Cura/files/13588541/CFFFNOTL_extruder_testcase.zip)

anubis71x, 2024-08-13:
> Hey @JAM3SY,
>
> Quick update from our side 👋 We are cleaning up our open issues so that we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I [...]

All of my profiles gave me this same behavior as long as enable bridge settings was turned on.

A recent post on reddit claimed this was the intended behavior according to a comment in 5.8.

"I found the following comment in the source code (cura 5.8)

if the proportion of the skin region that is supported is less than supportThreshold, it's considered a bridge and we determine the best angle for the skin lines - the current heuristic is that the skin lines should be parallel to the direction of the skin area's longest unsupported edge - if the skin has no unsupported edges, we fall through to the original code."

If this is true (I don't know if it is), then the "current heuristic" needs to be perpendicular to the direction of the skin area's longest unsupported edge.

https://www.reddit.com/r/Cura/comments/w002b4/change_bridging_direction_cura_making_worst/

JAM3SY, 2024-08-13:
Hi, Thnx for your continued support in chasing this problem. This anomaly was so long ago, I don't have the model anymore. Since buying bambu machine with an AMS, I've switched to Orca slicer. The kicker is... it happens on Orca too sometimes 😂 Especially where there are 2 long bridges running perpendicular to each other... Similar to the letter 'L'. Sometimes a diagonal bridge will form, other times it will not.  I usually circumvent it with ribs or webs in the model.

I haven't used Cura now since version 5.2.1, Not because I don't like Cura, simply because Orca handles color changes and multi-part models way easier.

Kind Regards
Brett

https://www.reddit.com/r/Cura/comments/w002b4/change_bridging_direction_cura_making_worst/