Ultimaker/Cura #10377: gCode using 2nd extruder which is disabled.

tlhintoq, 2021-09-01:
### Application Version

4.11.0b

### Platform

Windows 10 x64

### Printer

Hictop D3Hero

### Reproduction steps

Extruder 2 is disabled.
But profile has Extruder 2 for supports.
One would expect Cura to automatically shift calls from a disabled extruder to the first ENabled extruder.  IE: If extruder 2 is disabled then send the supports to 1.
If I have to enable T1 just so it appears in the "Supports Extruder" drop down. so I can then manually pick T0 - then what's the point of having a disable capability?
[image]

### Actual results

Extruder 2 is disabled.
But profile has Extruder 2 for supports.
One would expect Cura to automatically shift calls from a disabled extruder to the first ENabled extruder.  IE: If extruder 2 is disabled then send the supports to 1.
If I have to enable T1 just so it appears in the "Supports Extruder" drop down. so I can then manually pick T0 - then what's the point of having a disable capability?

### Expected results

Extruder 2 is disabled.
But profile has Extruder 2 for supports.
One would expect Cura to automatically shift calls from a disabled extruder to the first ENabled extruder.  IE: If extruder 2 is disabled then send the supports to 1.
If I have to enable T1 just so it appears in the "Supports Extruder" drop down. so I can then manually pick T0 - then what's the point of having a disable capability?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Extruder 2 is disabled.
But profile has Extruder 2 for supports.
One would expect Cura to automatically shift calls from a disabled extruder to the first ENabled extruder.  IE: If extruder 2 is disabled then send the supports to 1.
If I have to enable T1 just so it appears in the "Supports Extruder" drop down. so I can then manually pick T0 - then what's the point of having a disable capability?

fvrmr, 2021-09-02:
Hi @tlhintoq thank you for your report. I can't reproduce your issue, I get the following message, when disabling extruder 2 if it is set to be the support extruder. 

[image]

Could you share a project file (file - save project) then I can take a look. 
Thank you in advance.

Ghostkeeper, 2021-09-10:
I know this one. I've seen it myself as well. But I can't find it being tracked anywhere, here on Github nor on Ultimaker's internal issue tracker.

The reproduction steps provided by Tlhintoq are a bit unclear since they are not steps. These are the reproduction steps for this bug:
1. Load any multi-extrusion printer, like Ultimaker S5.
2. To see the difference in layer view, it helps to change the material in the second extruder to a different colour.
3. Enable support and set the support extruder to be the second extruder.
4. Save this as a custom profile.
5. Switch back to the default profile.
6. Disable the second extruder.
7. Load a model that needs support.
8. Enable support and slice. Inspect layer view in Material Colour mode.

You'll see that the support gets printed with the disabled extruder.

tlhintoq, 2021-09-10:
Thanks for that @Ghostkeeper 
I didn't even see the earlier request for more detail.  I guess I 'over pruned' my outlook email.
Yep, that certainly looks like the steps of how it happens to me.

Ghostkeeper, 2021-09-14:
Developers, see issue CURA-8568.