Ultimaker/Cura #10378: Pause at height Post Processing script pauses after layer instead of before layer

agibson2, 2021-09-02:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Duplicator 9

### Reproduction steps

1. Enable Post Processing 'Pause at height' script and set 'Pause at' to 'Layer Number', set 'Pause Layer' to 4, and set 'Method' to 'Repetier (@

agibson2, 2021-09-02:
This pic shows the third layer complete which is the default if you don't mess with the play button for that layer.  If you plug 3 into the Pause layer script, it would pause right after this layer.  Maybe the intent is for this script to pause after the layer # that you set because the expectation is that the user would see where they want to place the pause at and then use the layer number that they see for the value (which would be after the 3rd layer in this example).
[image]

In the example below, layer 4 is completed and someone might think that is too far because the GUI shows that the 4th layer is printed. so they might go back to the previous view and use than number (3)... not really thinking about what specific layer has or hasn't been printed... they might just go by what looks right and not actually counting layers.
[image]

This gets very confusing the express in words... hopefully someone can follow me :).

Asterchades, 2021-09-02:
Internally, Cura seems to refer to the first layer as layer 0. This is in contrast to the UI which shows it as layer 1. So I would suspect it's a disconnect between the way the layers are stored and the way it's displayed - telling it to pause at layer 3 would pause at the start of the _internal_ layer 3, which is actually layer 4 in the UI. Effectively this is the same as pausing at the end of the layer with a little bit of semantics as to how it's added.

I don't know if this is deliberate. But at least it is consistent enough that if you're aware of it you can plan for it.

fvrmr, 2021-09-02:
Hi @agibson2 thank you for your bug report. 
What @Asterchades mentioned is correct but it is also a bug we had before and for this bug we have a ticket on our backlog.
Devs see CURA-7443

agibson2, 2021-09-02:
I was maintaining my own octoprint puase script that I posted on the forums (adapted from one of the other included scripts) before the consolidated 'Pause at height' script was made recently so that is why I noticed it when I started using the new consolidated script.  I fixed it in my maintained script by subtracting 1 from whatever the user selected in the pause script :).  I thought I had screwed up when adapting it somehow but I guess I was 'fixing' it.

I suspect changing it now could upset a lot of people that are used to it working the 'wrong' way.

Ghostkeeper, 2021-09-13:
We had known about this issue before, filed under https://github.com/Ultimaker/Cura/issues/5085 . However for some reason that ended up a 404 link. Which is weird. As far as I know, no Github user has the ability to cause issues to disappear like that. We can only close them, not make them completely disappear.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

And it seems like #5085 is accessible again, so I will close this issue as a duplicate. Since that issue was also deferred.