Ultimaker/Cura #10379: Strange preview issues with knurled container

achampion, 2021-09-02:
### Application Version

4.10.0

### Platform

Mac Big Sure

### Printer

Snapmaker 2

### Reproduction steps

1. Load knurled model: https://www.thingiverse.com/thing:3175800
2. Slice with default settings
3. Preview model

### Actual results

This is the Preview image: 
[image]

### Expected results

A Preview without all the artifacts.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info.

achampion, 2021-09-02:
A reboot seems to have fixed the issue, please ignore.

closed 2021-09-02 by achampion

Ghostkeeper, 2021-09-13:
We've seen this before. The current going theory is that with Cura using a geometry shader, we're touching something that not a lot of applications do, and there seem to be bugs in some graphics drivers there as a result.
Looks cool though.