Ultimaker/Cura #10380: Profile Ultimaker 2 on cura is missing material customisation

leo33700, 2021-09-02:
### Application Version

4.10

### Platform

Windows 10

### Printer

Ultimaker 2

### Reproduction steps

There is no material and there is no possibility to modify or create any material with the preset ultimaker 2 profile.
[image]

And when I switch to other preset profile like ultimaker original it is possible
[image]

### Actual results

Nothing happened

### Expected results

There should be materials

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none
[image]
[image]

nallath, 2021-09-02:
That is by design; Materials are handled on the printer by the Ultimaker 2. 

We thought back then that that was a better way of doing it, but we realized that we were wrong, so future machines did have a combination of both.

closed 2021-09-02 by nallath

leo33700, 2021-09-03:
> 
> 
> That is by design; Materials are handled on the printer by the Ultimaker 2.
> 
> We thought back then that that was a better way of doing it, but we realized that we were wrong, so future machines did have a combination of both.

There was a time when i had this function on cura (i don't remember the version). And the ultimaker original is older than the um2 and it has the function on cura. It's just i like to see an approximation of the cost for an impression...

nallath, 2021-09-03:
> There was a time when i had this function on cura (i don't remember the version). 

It didn't. It could be that you set the g-code flavor to something different. The UM2 not having material profiles is because of the specific flavor of g-code that it requires. 

> And the ultimaker original is older than the um2 and it has the function on cura. It's just i like to see an approximation of the cost for an impression...

That is true. Like i explained, we changed the way it worked for the UM2, but we realised that this was not the best way forward (and thus switched back again for the UM3). Unfortunately, it has to do with the way that firmware handles these things, so it's not up to Cura to do this.

leo33700, 2021-09-03:
> It didn't. It could be that you set the g-code flavor to something different. The UM2 not having material profiles is because of the specific flavor of g-code that it requires.
>
>
> That is true. Like [...]

Ok and there won't be any update of the firmware of the UM2?

nallath, 2021-09-03:
> Ok and there won't be any update of the firmware of the UM2?

No. We did change this behavior for the UM2+ though (so for the UM2+ you can select materials in Cura)

fieldOfView, 2021-09-03:
What you can do is go into the Machine Settings in Cura, and change the gcode flavor to "Marlin" (instead of "Ultimaker 2"). This will let you set all material properties in Cura, but your printer will "complain" that the gcode will override any material properties selected on the printer.

leo33700, 2021-09-03:
> 
> 
> What you can do is go into the Machine Settings in Cura, and change the gcode flavor to "Marlin" (instead of "Ultimaker 2"). This will let you set all material properties in Cura, but your printer will "complain" that the gcode will override any material properties selected on the printer.

Thanks a lot this work like you said !