Ultimaker/Cura #10382: Cura Crashes on open file

TheLawlessOne, 2021-09-03:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creality CR-6SE

### Reproduction steps

1. open cura
2. Click on open file on left of printer

### Actual results

Cura closes down after briefly showing files

### Expected results

allow selecting STL or another file for inserting into CURA. Works fine when dragging file into the program 
[cura.log](https://github.com/Ultimaker/Cura/files/7107141/cura.log)
 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7107146/cura.log)

Mejituu, 2021-09-04:
我也是一样的

nallath, 2021-09-06:
Please check https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

closed 2021-09-06 by nallath