Ultimaker/Cura #10384: Increased line width only increases distance between lines

Salabok, 2021-09-03:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creality Ender-3

### Reproduction steps

1. Open the attached stl file with cura 4.10.0
3. Choose "Standard Quality" default profile
4. Change "Top/Bottom Line Width" setting from 0.4 to 0.6
5. Slice

### Actual results

The "Preview" tab shows the expected results, but when opening it with a gcode viewer (other than cura, because it shows the same as it shows in "Preview") we see that the skin lines don't touch, as if they remained 0.4 wide, and only the distance grew between them. Printing the object also shows the problem.

### Expected results

Skin lines that properly cover the surfaces

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Distant LInes.zip](https://github.com/Ultimaker/Cura/files/7108466/Distant.LInes.zip)

GregValiant, 2021-09-04:
Yes that's exactly right.  The line width is the index distance between adjacent extrusions.
Going from .4 to .6 is an increase of 50% and so your extrusion E values in the gcode should be 50% more for each individual extrusion.
At .4 line width you should see E values equivalent to 1mm of filament = 30.06mm of extrusion.
At .6 line width you should see E values equivalent to 1mm of filament  = **20.04mm** of extrusion.
In other words 30mm of .4 extrusion takes 1mm of filament and 30mm of .6 extrusion takes 1.5mm of filament.
From a gcode of your model with Skin line width at .6 (and sliced with Relative Extrusion so it's easier to read) these lines are in the first section of skin:
G0 F7200 X63.461 Y129.154
G1 F3000 X94.829 Y97.786 E2.21318
The above extrusion is 44.361 long and Cura gave it 2.21318mm of filament.  An E value of 2.21318mm of filament * 20.04mm of extrusion/mm of filament = 44.355mm of theoretical extrusion which is pretty close.  You should see ratios like that for every 0.6 wide extrusion in the gcode file.
I used MS Excel and AutoCAD to look at this to (take Cura out of the loop) but the Gcode Reader in Cura is separate from the Cura engine so the left hand doesn't know what the right hand did - it just reads the file in.
From the Gcode analyzer I used on the gcode file that I generated (from your model) I get this:

"Printed Line Length per mm of Raw Filament = **20.58 mm**".  Again, that's pretty close and the difference may be in "overlap" or something.

closed 2021-09-04 by Salabok

Salabok, 2021-09-04:
Well explained, the gcode viewer of simplify3D deceived me, it aligned perfectly with the badly printed object. But it turned out my printer was at fault.

GregValiant, 2021-09-04:
Thank you and thanks for getting back.  Sometimes people disappear and I never know if they were helped or if I was just confusing.