Ultimaker/Cura #10385: Feature Suggestion: Adaptive layer height gradual fan speed settings

RaptorTheOne, 2021-09-04:
### Is your feature request related to a problem?

I am a user that always has the "adaptive layer height" feature enabled.
Here are my settings:

> [v] Use Adaptive Layers
> [0.08mm] Adaptive Layerss Maximum Variation
> [0.04mm] Adaptive Layerss Variation Step Size
> [0.16mm] Adaptive Layerss Topography Size

What these setting above does:
-The thinnest possible layer height in adaptive layer height is 0.08mm
-The thickest possible layer height in adaptive layer height is 0.24mm
-The median layer height is 0.016mm
-It changes layer heights in 0.04mm stepping
-Transitions (Or differences in the next layer height) will not exceed 0.08mm difference

The problem I have without this potential feature is that at thinner layer heights (especially 0.08mm) (or even during the ironing process), the part cooling fan needs to be slowed down (or completely off during the ironing process)

[image]
This is an image of how I usually slice for high detail items. The blue layers shown here are 0.08mm layers.

[image]
For me, this is an image of the bottom side of the item printed 9first layer is always 0.2mm. This and the next few thicker layers printed just fine because the part cooling is working as intended.

[image]
This is an image of the top layers (shown in Acura as blue layers). It under extrudes because the cooling fan is overpowering the print flow for such a thin layer.

### Describe the solution you'd like

As the title suggests:
Settings like:

> [ ] Adaptive Layer Height Adaptive Part Cooling
>     [0.12mm] Layer height Part Cooling Layer Thickness threshold 
>     [50.0%] Threshold Maximum Part Cooling Fan Speed
>     [0.0%] Adaptive Layer Height Minimum Fan Speed
> [0.0%] Ironing Fan Speed

### Describe alternatives you've considered

Once completing the slicing operation:
-Manually find and write down the "thin" or "ironing" layers.
-Go to "Extensions" --> "Post Processing" --> "Modify G-Code"
-Choose: "Change at Z"
-Enter the affected layer height and adjust to desired part cooling fan speed

### Affected users and/or printers

All FDM printers with part cooling fans with the following settings enabled:

> [v] Use Adaptive Layers (with extremely small layer heights)
> [v] Enable Ironing

fvrmr, 2021-09-06:
Hi @RaptorTheOne thank you for your feature request. 
I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2021-09-09:
I have discussed it with the team. And we decided to reject this request. Since we don't think this would be the solution we to solve this issue. Hope you understand.

closed 2021-09-09 by fvrmr

Ghostkeeper, 2021-09-14:
Most settings need to change with the layer height. Temperature, overhang angle and line width being the most important ones. It could be fixed for the fan speed but in truth this will not result in a consistent print yet. This needs a re-think, especially since this suggestion would also significantly increase the complexity of settings.