Ultimaker/Cura #10386: Cura Crashes on open file,but found the problem.

Mejituu, 2021-09-04:
### Application Version

4.10.0

### Platform

Windows 10 or 11

### Printer

CoreXY 兔兔

### Reproduction steps

1. open cura.
2. Click on open file on left of printer.

### Actual results

Cura closes down after briefly showing files.

### Expected results

allow selecting STL or another file for inserting into CURA. Works fine when dragging file into the program
cura.log

I found from the error.log that the address of the request Python is C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\ this directory, but the actual user of my computer The name address is C:\Users\懵仙兔兔\AppData\Local

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/7109170/stderr.log)

Mejituu, 2021-09-04:
[image]

Mejituu, 2021-09-04:
我的电脑实际用户文件夹名是这个[image]

Mejituu, 2021-09-04:
I think this should be a bug since 4.9

Mejituu, 2021-09-04:
This is the cura-crashyttefy_l file in the C:\Users\懵仙兔兔\AppData\Local\Temp directory.
[cura-crashyttefy_l.log](https://github.com/Ultimaker/Cura/files/7109196/cura-crashyttefy_l.log)

fieldOfView, 2021-09-05:
Do you have chitubox installed on the same system? If so, see this article:
https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

Mejituu, 2021-09-05:
> 在同一系统上安装了气盒吗？如果是这样，请参阅
> 本文[：https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file](https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file)

我的电脑没有安装过 Chitubox

Mejituu, 2021-09-05:
> Do you have chitubox installed on the same system? If so, see this article:
> https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

My computer has not installed Chitubox.

Mejituu, 2021-09-05:
2021-09-05 18:48:19,546 - CRITICAL - [JobQueueWorker [1]] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2021-09-05 18:48:19,550 - CRITICAL - [JobQueueWorker [1]] cura.CrashHandler.__init__ [69]: AttributeError: 'NoneType' object has no attribute 'close'
Exception ignored in: 'scipy.spatial.qhull._Qhull.__dealloc__'
Traceback (most recent call last):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 512, in createConvexHull
AttributeError: 'NoneType' object has no attribute 'close'
2021-09-05 18:48:19,795 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [112]: Exception: Failed to open file b'C:\\Users\\\xe6\x87\xb5\xe4\xbb\x99\xe5\x85\x94\xe5\x85\x94\\AppData\\Local\\Temp\\scipy-_u5_y1sc'
2021-09-05 18:48:19,800 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]: Traceback (most recent call last):
2021-09-05 18:48:19,802 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshFileHandler.py", line 45, in readerRead
2021-09-05 18:48:19,805 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 220, in getExtents
2021-09-05 18:48:19,808 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 296, in getConvexHullVertices
Windows fatal exception: access violation

fvrmr, 2021-09-06:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9644
The temp folder doesn't have access because of non-Latin characters in your Windows username. This should be fixed for 4.11 stable. 
I will close this as a duplicate.

closed 2021-09-06 by fvrmr