Ultimaker/Cura #10387: Slicer preview is nonsensical

shdwshard, 2021-09-04:
### Application Version

4.10.0

### Platform

Mac OS X

### Printer

Tronxy X5SA

### Reproduction steps

Tried to slice a large model
Visited the Preview screen

### Actual results

The visualization is nonsensical with "rays" eminating from the model, and the model itself is not rendered properly

### Expected results

The visualization should have been an accurate representation of the slicing results

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Garden Tank.3mf.zip](https://github.com/Ultimaker/Cura/files/7109641/Garden.Tank.3mf.zip)
[image]

fvrmr, 2021-09-06:
Hi @shdwshard do you have a M1? If so, it is a known issue. This is happening because Cura uses OpenGL libraries and with the new Macbooks containing the M1 processing chips, Apple has introduced incompatibilities between the OpenGL libaries used and the latest firmware used by Macbooks.

mikelexp, 2021-09-08:
I'm having the same problem with 4.11 on an Intel Mac (MacBook Pro 16 i7). 4.10 was ok.

[image]

shdwshard, 2021-09-08:
No, I have an intel based MacBook Pro.

Interestingly a reboot solved the issue, so more than a direct incompatibility, it seems like the OpenGL libraries used by Cura are unstable on newer MacOS versions.

This is also a machine I use for gaming, so it's possible some state gets left over, or the is switches to the discrete GPU in a way that Cura doesn't like.

fvrmr, 2021-09-10:
@shdwshard good to know that a reboot solved your issue. It is correct that the OpenGL libraries are not working well with MacOS especially with M1. Are you maybe using Monterey Beta?

A possible work around is to set manually for `force_empty_shader_cache = True` under `[view]` in the cura.cfg file. If there is no [view] header you can create one. 
You can find your cura.cfg file via Help - Show Configuration Folder.

shdwshard, 2021-09-11:
No, I'm on Big Sur, and I will definitely try that if it happens again.

Apple seems to be really focused on Metal, so I guess it's no surprise that OpenGL compatibility is sliding/suffering in recent OS releases.

Just FYI, here are some more detailed stats of my system so you have another data point as to what systems can hit this problem.
```
MacOS Big Sur
Version 11.5.2 (20G95)
MacBook Pro (16-inch, 2019)
Processor 2.3 GHz 8-Core Intel Core i9
Memory 16 GB 2667 MHz DDR4
Graphics AMD Radeon Pro 5500M 4 GB
```

Ghostkeeper, 2021-09-14:
So... I'm guessing this is really just a bug in the OpenGL implementation of your video drivers?

We are kind of stuck with Qt, and Qt doesn't support Metal. It does support Vulkan, but this will require some extra tricks to pull off since Vulkan libraries are not provided by MacOS itself; we'd need to provide it with our distribution and I don't know how portable that would be with the 4 versions of MacOS that Cura currently supports. And I fear that Vulkan is going to be in a similar state to OpenGL, perhaps even worse because it's less battle-tested than OpenGL.

I don't know what we at Cura could really do about this problem.

antoooooooooooonie, 2023-05-09:
Confirmed to still happen on Cura 5.2.1

Not sure if yet another screenshot of a glitched preview is even useful at this point but:

[image]

Computer specs:

[image]

Is there any known fix for this besides downgrading Cura? Or is it related to me running macOS Ventura?

shdwshard, 2023-05-09:
Shutting down and restarting the machine fixes the issue for me, and tends to make me think this is a GPU or GPU Driver state issue more than a version specific problem.  My M1 has some other weird transients with graphics overlays and such which are also remedied by a cold boot.

altimac, 2023-08-24:
> @shdwshard good to know that a reboot solved your issue. It is correct that the OpenGL libraries are not working well with MacOS especially with M1. Are you maybe using Monterey Beta?
>
> A possible [...]

Unfortunately I can confirm that this hack does not work. At least once the problem appeared, quitting Cura,editing the cura.cfg file to add this setting and relaunching cura (reopening the same project), slicing and previewing still shows the same bug.

GregValiant, 2024-11-16:
The "Peter Maxx" bug.
This has never actually been fixed and every once in a while a new bug report is made.