Ultimaker/Cura #10389: M600 Marlin filament change makes blob if retract between layers

ceandraka, 2021-09-04:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Prusa i3

### Reproduction steps

Using Prusa i3 settings on a P3Steel with Marlin 2.0.1.

Using Cura Post Processing Script "Filament Change", select layer (118), check "Use Firmware Configuration", and z Position 5mm. Also, "enable retraction" is checked, with length 6.5mm. Resulting gcode attached

[PI3_Protractor Stand 195C 2color.zip](https://github.com/Ultimaker/Cura/files/7110391/PI3_Protractor.Stand.195C.2color.zip)

### Actual results

Print proceeds fine until filament change. Filament change uses M600, which proceeds fine, and ends with a purge, and then approval to continue (or purge more). Then head returns to print position, and initially spurts out a blob of filament about 3-4mm diameter.

See lines 47195-47217

47195: Final extrusion of first color
47197: Retract 6.5mm for layer change
47213: M600 filament change
47217: Un-retract to get ready for print

I believe that the purge process (M600) leaves the extruder full rather than retracting 6.5mm (the firmware routine has no way of knowing about the 6.5mm retraction that was done), and then line 47217, un-retract, expels 6.5mm of filament onto the print

### Expected results

Some sort of coordination between the retract and the M600 is needed. M600 appears to return the extruder to the prior position, ready to print. However, "prior position" includes a 6.5mm retraction, so it picks up with a 6.5mm un-retraction. Cura "knows" that this retraction has occurred, and knows it will unretract. 

Expected result would be to NOT unretract, or to do a relative retraction after M600 to restore prior position, and then proceed with absolute un-retraction.

Short-term workaround: Manually remove the retraction statement on line 47197, risking stringing but that would make the un-retraction on 47217 do nothing.
[PI3_Protractor Stand 2-color with issue.zip](https://github.com/Ultimaker/Cura/files/7110403/PI3_Protractor.Stand.2-color.with.issue.zip)

Project file zipped and attached.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Project file and gcode are zipped and attached

fvrmr, 2021-09-09:
Hi @ceandraka thank you for your report. 
Could you try to unable the use firmware configurations in the post processing script and look if it doesn't happen then.

ceandraka, 2021-09-10:
I ran a smaller test article with color change. I ran it twice, both times without using printer settings (In plugin, Use Firmware Configuration is not checked). In the first case, I have 6.5mm retraction, and check retract between layers. This case acts the same as before, a big blob at the start of the second color. The second case I do not check retract between layers. There is no blob.

The only difference in the created gcode is the retraction before, and anti retraction after, the M600 command.

I used the default values in Cura for the filament change operation, which results in: M600 E30.00 L300.00 X0.00 Y0.00 Z5.00

With retraction, the observed printer action is:
-Complete prior layer
-Retract 6.5mm
-Retract 30mm (part of M600 command)
-Move to filament change location
-Retract a bunch
-Insert new filament and auto purge (no additional retract is observed at end of purge)
-Acknowledge purge is complete (no additional retract is observed)(So now filament is at tip of nozzle, not retracted)
-Move back to part, lower head
-Un-retract 6.5mm, creating a blob

On the method that works (no retraction< the observations are the same, except:
-No 6.5mm retraction at the end of the layer
-No 6.5mm un-retraction at the start of the layer

The M600 command, which manages the 30mm retraction  and 300mm load length to perform filament change, has no way of knowing that the layer control has retracted 6.5mm, and therefore no way of knowing to leave 6.5mm space in the extruder head.

Again, current workaround is to remove the retraction command in the gcode file just before the M600 command, or to turn off retraction between layers.

Including
[FilamentChangeSample.zip](https://github.com/Ultimaker/Cura/files/7145766/FilamentChangeSample.zip)
 new gcode and project files

Ghostkeeper, 2021-09-14:
Yeah, that looks like a bug. The post-processing script simply inserts the `M600` command (with parameters or without, depending on if it's supposed to use firmware defaults). It doesn't calculate if the material is currently retracted or not, or how far it is retracted.

The `M600` command is expected to do all of the priming necessary to start printing (firmware dependent), so it'll end up with the material in the nozzle tip. The unretract following then is going to purge material, which would end up on the print.

Ghostkeeper, 2021-09-15:
This was discussed with the team, but considering our other priorities, how we don't have a printer that can run the M600 command, and how rarely this gets used in the field, we've decided that we'll have to defer this ticket to another time. It's an issue so we'll keep it open, but not on our priority list now.

ceandraka, 2021-09-15:
OK, thanks for looking. I will manually remove the retract line before the M600, or I will discontinue using retract between layers.

Chuck

ceandraka, 2023-05-26:
I just downloaded and installed v5.3.1, and ran the same sample code.

I have verified the problem still exists in this version.

Currently the best workaround is to add an unretract immediately before the M600 command. At this point, however, the nozzle is still over the part and could extrude some filament onto the part. Alternatively, the retract just before the M600 command can be removed. Either way, the result is a full nozzle right before the M600 command, which is the state that the M600 command leaves things after filament change.

Thank You
Chuck

MariMakes, 2023-06-05:
Hey @ceandraka,

👋 A fresh set of eyes reporting for duty! 💪 
Thank you so much for testing and sharing the results. 

This issue strongly reminds me of this report from two weeks ago: https://github.com/Ultimaker/Cura/issues/15564
The report mentions that filament was not retracted for a filament change related to an m600 command, we removed some `/n`s in the post-processing script to resolve the issue.

Can you check if the workarounds mentioned int he ticket help you as well?
If you still run into the issue feel free to let us know.
I'll take another look at what is going wrong. 👍

ceandraka, 2023-06-05:
This is not the same issue that I saw. In my cases, the M600 command was properly formed.

My issue is when the run is set to retract between layers. It does a retract at the end of the first color last layer, then issues M600 command. This causes the filament change, the end result of which includes purging, which fills the nozzle. When control returns from the M600 command, the assumed filament position is retracted (last state before M600), but in fact the M600 refilled the nozzle. Therefore, Cura code advances the filament (unretract), resulting in a blob.

This is ONLY an issue if retraction between layers is used along with a filament change.

The best workaround at this point is to remove the filament retraction just prior to the M600 command.

Thanks for looking

Chuck

MariMakes, 2023-06-14:
Hey @ceandraka,

I brought up your issue with the team.
They reminded me that you can define how the M600 is being inserted because there are different types of firmware that handle retracts differently.
[image]

Does the Use Firmware Configuration checkmark give you the behavior you are looking for? 

We also believe we will not work on this anytime soon and will therefore close this issue. Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself. I hope you understand. 

>

closed 2023-06-14 by MariMakes