Ultimaker/Cura #10390: Side Bar - Print Configurations not resize with mouse cursor

MarcusVieira01, 2021-09-05:
### Application Version

4.11.0 Beta

### Platform

Windows 10

### Printer

Crealitu Ender 5 Pro

### Reproduction steps

1. Click and hold with left button on de resize on botton print configuration bar
2. Move the mouse to resize
3. Nothing happens

### Actual results

Wrote this report

### Expected results

Resize bar 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing extra

fvrmr, 2021-09-06:
Hi @MarcusVieira01 could you post a screenshot of what you exactly mean? I am not sure if I understand your issue correctly.

MarcusVieira01, 2021-09-07:
Hi Fenne,

Thank for your return. I apologize for my bad english.

After mailing you, I was successful to resize the configuration bar, in
right side

[image: image.png]

I beggin pardon for sanding the message with few tests.

Again, thank you for the return.

Greetings.

Marcus Vinícius Magdaleno Vieira
+55 11 970261069

fvrmr, 2021-09-09:
Thank you for letting me know it works! I will close this issue since it is solved.

closed 2021-09-09 by fvrmr