Ultimaker/Cura #10391: gcode file corrupted while transfer to Octoprint

kanawati975, 2021-09-05:
### Application Version

4.11.0

### Platform

Windows 10 

### Printer

DIY

### Reproduction steps

After slicing the file and sending it to Octoprint. The print Job starts normally and at some point the printer hangs and Octoprint reports an Error

### Actual results

I downloaded the gcode and opened it with notepad++ to see a huge portion of the file contains gibbrish.

### Expected results

This occurs only when using "send to Octoprint" button.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2021-09-05:
If this is an issue with the OctoPrint Connection plugin, please open an issue there instead:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues

Having said that, I don't see how the OctoPrint Connection plugin can do this. All it does is send the gcode file through the REST API that the OctoPrint web interface itself uses too. Is this reproducable on your system? As in: does it happen all the time?

kanawati975, 2021-09-05:
Not always. But it's like 2 out of 5 attempts.
Uploading gcode files directly to Octoprint is fine.

fieldOfView, 2021-09-05:
Very strange. But like I said, please open an issue over at the plugin repository, instead of here.
Include a link to your Cura.log, and OctoPrint log files.

kanawati975, 2021-09-05:
I just did.
Thank you for your recommendation.
Should I close this one?

fieldOfView, 2021-09-05:
Yes please.

closed 2021-09-05 by kanawati975