Ultimaker/Cura #10392: Some of the empty interiors of the model are filled during slicing.

zorgoz, 2021-09-05:
### Application Version

4.10

### Platform

Window 10

### Printer

Ender3 V2

### Reproduction steps

1. Download this model: https://www.thingiverse.com/thing:453672
2. Choose any preset and slice.  

### Actual results

Some of the interiors of the model are filled.

### Expected results

The model is a cookie-cutter, it is merely walls, as SLIC3R is slicing it.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The model is reported not being watertight, but it looks ok.
[creeper_slicing_bug.zip](https://github.com/Ultimaker/Cura/files/7111663/creeper_slicing_bug.zip)

GregValiant, 2021-09-05:
I think it's the way the model was created.  In Cura go to the Mesh Fixes section and turn off "Union Overlapping Volumes".

[image]

Ghostkeeper, 2021-09-14:
Indeed, as seen here in CloudCompare, which renders the inside surface blue and the outside surface green:
[image]

The inside of the creeper face has its surface inside-out. So Cura thinks they are just columns of doubly filled material inside of the creeper face. The inside (hollow) shape overlaps with the outside of the creeper face, and since Union Overlapping Volumes is enabled it combines the overlapping shapes into one single (filled) shape. With Union Overlapping Volumes disabled, the [even-odd](https://en.wikipedia.org/wiki/Even-odd_rule) fill rule is used instead, causing Cura to make the overlap empty and not sensitive to the direction of the surface any more.

So this is indeed a problem with the model.

The MeshTools plug-in from the Marketplace has a workaround. There is an option to attempt to fix the direction of the surface (normals). It's not 100% reliable but works well for most cases.

closed 2021-09-14 by Ghostkeeper