Ultimaker/Cura #10393: Intent profile is displayed even though the nozzle of the right extruder is deactivated.

x40-Community, 2021-09-05:
### Application Version

Master

### Platform

Linux

### Printer

Weedo X40 / Idex printer

### Reproduction steps

Open a project file from a dual extruder printer with the following setup:
The extruder on the left is set to a 0.4mm nozzle. The left extruder is set to a nozzle 0.6mm and deactivated. 

### Actual results

The intent profiles of the 0.6mm nozzle are displayed, although the nozzle is deactivated. 

### Expected results

If the nozzle is or will be deactivated, the associated profiles should no longer be displayed. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fvrmr, 2021-09-06:
Hi @x40-Community could you share a project file? It is easier to reproduce your issue with a project file.

Ghostkeeper, 2021-09-14:
Your issue can't be reproduced in our Master version because we don't have intent profiles yet with the Weedo X40 printer. A project file wouldn't be sufficient then; we'd need the intent profiles too (and any other profiles you may have changed).

It seems to be behaving fine as far as I can see in the Ultimaker S5. Here I put PLA on the right extruder which has intent profiles, but PP on the left extruder which doesn't have intent profiles. Disabling the right extruder causes the intent profiles to disappear. The horizontal slider remains unchanged because both extruders support the same quality levels in this case. The drop-down with profiles reflects this too; it is the same set of profiles, in both cases, as the sliders in Recommended Mode. Recommended Mode visualises it a bit nicer though.
[image]
[intent_profiles_disabling_extruders_s5.zip](https://github.com/Ultimaker/Cura/files/7161491/intent_profiles_disabling_extruders_s5.zip)

Ghostkeeper, 2021-09-14:
Ah, the intent profiles seem to be the ones provided here? https://github.com/Ultimaker/Cura/pull/10345

x40-Community, 2021-09-14:
The quality and intent profiles are only just gradually being added to Cura.

x40-Community, 2021-09-14:
[bug.zip](https://github.com/Ultimaker/Cura/files/7163997/bug.zip)
Here is the project file for the error

Ghostkeeper, 2021-09-21:
> [bug.zip](https://github.com/Ultimaker/Cura/files/7163997/bug.zip)
> Here is the project file for the error

Thanks! But this zip archive is corrupt? File Unroller can't read it in any case :(

> Do you always have to create the profiles for all nozzle sizes, even if they don't work with the printer?

No, preferably you'd only have profiles that work.

These are the relevant requirements we have:
* The available quality levels is the *intersection* of the quality levels available for each individual (enabled) extruder. For instance, if one extruder has quality levels Fine and Normal, and the other extruder has quality levels Normal, Draft and Sprint, then the user can only select Normal, since that is the only quality type available for all extruders.
* The available intent categories is the *union* of the intent categories available for each individual (enabled) extruder. So if an intent profile is available for any extruder, the user can select it. If a different extruder doesn't have an intent profile with that intent category, the intent profile will be left empty (in the interface listed as "default").

closed 2024-12-09 by github-actions[bot]