Ultimaker/Cura #10394: Single Layer print fails to slice with current thickness settings

lwsmiser, 2021-09-05:
### Application Version

4.11

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

1. Opened Cura 4.11 beta
2. Opened First_Layer_Test_.2.stl
3. Selected profil
4. Sliced loaded file

### Actual results

[image]

### Expected results

Expected the file to be sliced

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[First_Layer_Test_.2.zip](https://github.com/Ultimaker/Cura/files/7112181/First_Layer_Test_.2.zip)

lwsmiser, 2021-09-05:
The file will slice if I change the Z dimension to 0.2006 but will not slice if I leave the Z dimension at 0.198 and change the first layer height to 0.201 (Cannot enter 0.2006 here.)
[image]
[image]

fvrmr, 2021-09-06:
Hi @lwsmiser could you also share your project file (file - save project)? It is easier to reproduce your issue with a project file.

lwsmiser, 2021-09-07:
[First_Layer_Test_.2.zip](https://github.com/Ultimaker/Cura/files/7118739/First_Layer_Test_.2.zip)

fvrmr, 2021-09-10:
Thank you. I will bring this up with the team. Keep you posted!

hitboxr, 2021-09-11:
I just tested and the same thing happens in 4.10, so this doesn't seem like something that was introduced in 4.11

smartavionics, 2021-09-11:
See https://github.com/Ultimaker/Cura/issues/10425

Ghostkeeper, 2021-09-14:
> [First_Layer_Test_.2.zip](https://github.com/Ultimaker/Cura/files/7118739/First_Layer_Test_.2.zip)

This archive appears to be corrupted? I'm unable to open it.

lwsmiser, 2021-09-15:
The file is a 7 zip file.  I had to change the extension to upload it to your page.  Try changing the extension to .7z  .
I just downloaded it from the link in your email and it opened correctly.

lwsmiser, 2021-09-15:
Here is the link to the file in Thingiverse.
https://www.thingiverse.com/thing:2125379
Use the second file: First_Layer_Test_.2.stll

I just checked the file in v 4.9.1 and v4.10   and found that it will open and slice only in 4.9.1 with a thickness of 0.2 mm but in the higher versions it will not slice and the thickness is shown as .198 mm instead of 0.20 mm.

Ghostkeeper, 2021-09-17:
I made an analysis of this bug here: https://old.reddit.com/r/Cura/comments/povb83/cura_411_does_not_slice_bed_leveling_files_anymore/hd3h73n/

The check where it sets the layer count to 0 is wrong.

It should only set it to 0 if:

* the slicing tolerance is "Middle" and the print is lower than half the initial layer height, or
* the slicing tolerance is "Exclusive" and the print is lower than the initial layer height, or
* the slicing tolerance is "Inclusive" and the print has height 0.

konskarm, 2021-10-05:
This issue is currently in our backlog CURA-8614.

nallath, 2021-10-20:
Fixed in https://github.com/Ultimaker/CuraEngine/pull/1512

This is included in the 4.12 release that will go into beta soon.

closed 2021-10-20 by nallath

richfelker, 2022-03-15:
The fix for this seems to have introduced a new bug: https://github.com/Ultimaker/CuraEngine/issues/1606