Ultimaker/Cura #10398: Cura won't slice the model

Enderdragon2236, 2021-09-06:
### Application Version

4.9.1

### Platform

Mac OS 11.5.2

### Printer

Ender 3

### Reproduction steps

1. I made a model in Tinkercad that was 210x210 millimeters (Ender 3 build volume is 220X220)
2. I exported it to Cura
3. I tried to slice it and it said the model was too big for the build volume. What?!?

### Actual results

It said the model was too big and wouldn't slice

### Expected results

The model was just fine and Cura sliced it just fine

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

there is no additional info

Enderdragon2236, 2021-09-06:
Please help!!

nallath, 2021-09-06:
Please upload the model, preferably in a project file.

Ghostkeeper, 2021-09-14:
Perhaps you will need to reduce the width of the brim around your 210mm wide object? You only have 2.35mm of space around it, so your brim will need to be quite small.

Enderdragon2236, 2021-09-20:
Never mind it was because I left the bed adhesion on. Dumb noob error.

jellespijker, 2021-09-20:
Good to know it is fixed and don't worry we have all been dumb noobs at one point in our lives 😉

closed 2021-09-20 by jellespijker