Ultimaker/Cura #10399: Objects move across build plate when asking to lay flat. And objects that fit perfectly with optimal space gets pushed off the build plate, saying can't find a place for object.

stonemandave, 2021-09-06:
### Application Version

4.10

### Platform

windows 10

### Printer

Ender 5 Pro

### Reproduction steps

1. Objects move and rotate when just asking to lay flat.
2. Objects that fit with no problem manually gets pushed off with ever small item added.

### Actual results

Objects move for no reason. I have plenty of space on the buildplate your algarithm you use to detect where items can fit needs much improvement.

### Expected results

I expect items not to move when im not asking them to.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2021-09-09:
Hi @stonemandave can you share a screenshot of what is happening? And share the models where this is happening.

Ghostkeeper, 2021-09-14:
Yeah, we've known about this issue since Lay Flat was introduced. We didn't think it was worth worrying about much. Here is a video of the issue:
https://user-images.githubusercontent.com/2448634/133262078-a5365757-82fa-4135-93b1-11ab09aa518c.mp4

We'll leave this issue open for it to be tracked properly, but we don't consider it important enough to plan in our roadmap at the moment.

closed 2023-06-24 by github-actions[bot]