Ultimaker/Cura #10400: Hole not being sliced

reza0310, 2021-09-06:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality CR 10S pro

### Reproduction steps

1. I made the model using blender
2. I exported it as an STL file
3. I imported it in Cura
4. I scaled it down since the auto-scale was too much but I don't have to do this step with the creality slicer even if i have the same problem since the file is technically scaled properly
5. I sliced it and went to the layers view to check
PS: "Remove all holes" parameter is off and "Union overlapping volumes" parameter doesn't change anything but is currently off too
PS2: I thought it could be too precise for the printer but i already printed the same model with the hole even smaller and it worked. I didn't change any parameters so it might be my model or I just got lucky I don't know I'm a begginner.

### Actual results

The middle hole isn't sliced at all

### Expected results

I expected the layer view to have the hole that all the other views show

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[project.zip](https://github.com/Ultimaker/Cura/files/7117126/project.zip)
[image]
[image]
[image]

GregValiant, 2021-09-07:
When I turn off "Union Over-lapping Volumes" it's fine.
[image]
[GVproject.zip](https://github.com/Ultimaker/Cura/files/7123040/GVproject.zip)

reza0310, 2021-09-07:
The parameter is on each time I open Cura but deactivating it just before slicing work as you said. I might have deactivated it too late or thought it will still be deactivated whenever I reopen Cura and that's what blocked me probably. Thanks a lot I'm closing this issue.

closed 2021-09-07 by reza0310

GregValiant, 2021-09-07:
When you start Cura and that "Profile" loads make sure of all settings and then click on the Profile name.  Near the bottom of the list of profiles will be a command "Update profile with current settings".  That should keep the setting from loading "turned on".