Ultimaker/Cura #10401: Cura calling T0 for every layer even when not used

cbldwn, 2021-09-07:
### Application Version

4.10

### Platform

Windows 10

### Printer

Custom running Duet

### Reproduction steps

Slice any model that uses 4 different colors and only uses tool 0 in the first couple layers.

### Actual results

Calls up T0 on every layer

### Expected results

My model has 103 layers with T0 being used only on the first 31 layers. Cura continues to call T0 on every layer after 31 but only returns to the last position (no extrusion) before calling up the proper tool to continue printing. I've looked to see if there is any kind of wipe or anything and have been unable to find anything.  I've attached a copy of the gcode that is useless and causing problems with adding a ton of time and wasting a ton of filament with each tool change it does for no reason, the lines with the dashes are the ones that have no reason to exist in the first place. I have attached the project file as well. the issue starts on line 179968-179971 and repeats on every layer after even tho that tool is no longer required of doing anything.

;TYPE:SKIN
G1 F1500 X140.434 Y171.823 E0.00477
M566 X900 Y900
;MESH:NONMESH
G0 F7200 X140.355 Y171.902
G10
G1 F300 Z4.08
T0                                                               -----
;MESH:wood_-_4.stl                                    -----
G0 F7200 X140.355 Y171.902 Z4.08                                    -----                    
G1 F300 Z4.08                                    -----
T1
M106 S255
G10
G1 F300 Z3.96
G0 F7200 X154.008 Y137.002
M566 X540 Y540
;TYPE:WALL-INNER
G11

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[D_wood_-_1.zip](https://github.com/Ultimaker/Cura/files/7118053/D_wood_-_1.zip)

AndreasALu, 2021-09-08:
Could be connected / similar with this problem:

https://github.com/Ultimaker/Cura/issues/8769

fvrmr, 2021-09-09:
Hi @cbldwn it could indeed be related to #8769 like @AndreasALu mentioned.
I can't open your project file since I don't have your printer configurations. Do you have the print sequence set to one-at-a-time mode?

cbldwn, 2021-09-09:
No I don’t, I’m only printing a single multi color part and because I’m using more the 1 extruder (4) that setting is hidden.

AndreasALu, 2021-09-09:
I also can´t open your project file. How do you change the colours / how do you set up Cura so that it stops or does whatever is needed to change the colour?

cbldwn, 2021-09-09:
I’m running a Duet2WIFI so I have my tool changes configured on the printer so that when a tool is called it will move to x-16 y150 and retract the current filament back into the 4-way Y splitter I have installed in my Bowden tube. It will then advance the next tool/color and purge the previous color in my purge bucket and will auto wipe the nozzle when it continues the print.

AndreasALu, 2021-09-09:
I get that. But how do you set up the tool change **in Cura**?

cbldwn, 2021-09-09:
No setup in cura required for my tool changes, cura issues the tool call depending on which color is needed and my printer takes care of everything else. The settings that would normally go into each extruder in cura I have setup if my printer firmware so I can tweak the setting during a print sorta like how firmware retract works where cura issues a G10 and the firmware will handles the retraction.

fvrmr, 2021-09-10:
Could you share your printer configurations with us?

cbldwn, 2021-09-15:
[Duet Printer json files.zip](https://github.com/Ultimaker/Cura/files/7165775/Duet.Printer.json.files.zip)
Sorry was a busy weekend. Here are the json files I created for my printer, I've updated the M104/9 to G10 in my more recent configuration but haven't had a chance to update it in the json files yet. If anyone know of a way to completely remove T0 from cura and have it start ordering the tools starting at 1 instead of 0 so that I can set my printer to unload its current  tool when T0 is called I would be interested to know. As far as my original issue goes I found out that if I load what ever color finishes the print (dark green for the trees) in T0 then the tools get ordered as one would expect, No useless tool changes .

Ghostkeeper, 2021-09-15:
The project file also refers to definitions called `Duet_extruder_5` and `Duet_extruder_6` which weren't included. I can extrapolate from the existing files, but I won't know if it's the same case then. It's also saying that the nozzle profiles are missing.

When I run that, I'm getting a segfault in the `LayerPlanBuffer::addConnectingTravelMove` function (which calculates the path to the starting position of the layer). I'm getting this because I run CuraEngine in debug mode and it does some extra assertions then. The segfault happens because the first extruder plan of the layer is apparently empty. This might be the cause of the issues we're facing here.

cbldwn, 2021-09-15:
Extruder 5 and 6 are exactly the same as the first 4 extruders beside the name change for the extruder number. With my current setup I have set cura in the settings to have 4 extruder as the last 2 are for future additions. By missing nozzle profile are you referring to the nozzle diameter, nozzle position or something else?

Ghostkeeper, 2021-09-17:
The nozzle profiles. You'll see in the log file that it's expecting variant profiles but there are none.

Ghostkeeper, 2021-09-17:
Should be noted though that I did get something to fail even without those nozzle profiles, which is most likely the same bug as you or at least related to it.

cbldwn, 2021-09-19:
Where would the nozzle profiles normally be located?

Ghostkeeper, 2021-09-22:
In the `variants` subfolder of the configuration folder. They could be anywhere though. Cura doesn't care about which resource folder they are located in.

closed 2024-12-09 by github-actions[bot]