Ultimaker/Cura #10402: Select face to align to build plate does not work correctly.

BobCraig10, 2021-09-07:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

1 select "Select face to align to build plate"
2 Click on face you wish to have on the build plate

### Actual results

The part ends up at ridiculous angles on the build plate nowhere close to the face you chose and nowhere close to printable. I have to clear the build plate and reload to be able to do anything.

### Expected results

The face you chose to be on the build plate should have to been moved to the build plate.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2021-09-09:
Hi @BobCraig10 can you share your project file?
Or some screenshots of what exactly is happening?

BobCraig10, 2021-09-09:
I'm sorry, I updated to 4.11 yesterday, and uninstalled 4.10. and do not
have the issue with it.

fvrmr, 2021-09-09:
Thank you for the update! I will close this issue since it is solved in 4.11!

closed 2021-09-09 by fvrmr