Ultimaker/Cura #10407: Monotonic ironing causes fanspeed jump to 255

ferociousdiablo, 2021-09-07:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Add model
2. Enable monotonic ironing
3. Slice
4. Check gcode file for extra M106 commands. Found M106 S255 on layer 99 in attached project.

### Actual results

Fanspeed jumps to max when it should not on layer 99
[Cura fanspeed bug reproduction.zip](https://github.com/Ultimaker/Cura/files/7124404/Cura.fanspeed.bug.reproduction.zip)

### Expected results

Leave fanspeed as user has set it and do not interfere.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

First noticed this on a custom model which I cannot upload project for as it is an item I sell. Fan speed in that file actually jumps to max and then back to what it should be when the layer is finished. Happens on a seemingly random layer and several times in the custom model I was slicing but only once in the calibration cube project I have uploaded.

ferociousdiablo, 2021-09-07:
Also noticed that if you disable "Iron only highest layer" there is an additional M106 S255 on layer 25 which returns to specified speed on layer 26, and an erroneous / extra travel move on layer 26 (picture attached) which has been a long standing bug in Cura which I previously reported back in 4.6 or so. There is also an extra M106 S255 on layer 94 which returns to specified speed on layer 95 with only highest layer only disabled.

[image]

fvrmr, 2021-09-09:
Hi @ferociousdiablo thank you for your report. 
I will bring this up with the team. Keep you posted.

nallath, 2021-09-13:
I've found the issue. Fix is pending testing  & review.

ferociousdiablo, 2021-09-13:
Thanks for the quick fix on this! Was driving me mad until I noticed what was going on with my ABS prints lol

closed 2021-09-14 by Ghostkeeper