Ultimaker/Cura #10411: Cura app can't connect Anet A8

skabrits, 2021-09-08:
### Application Version

4.11.0

### Platform

Mac OS (Catalina)

### Printer

Anet A8

### Reproduction steps

Join printer to Mac by cable. Open Cura. Go to page Monitor.

### Actual results

No printer appears, white field with no interface (as if it was not connected).
[image]

### Expected results

The printer should be detected and standard monitor interface should appear.
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

To fix problem in file /Applications/Ultimaker Cura.app/Contents/Resources/plugins/plugins/USBPrinting/AutoDetectBaudJob.py line 77 should be changed from:
if b"ok" in line and b"T:" in line:
[image]
to:
if b"ok" in line or b"T:" in line:
[image]
As 3d printer send "ok" and "T:" on different lines, Cura can't detect it.

Unfortunately I can't attach fixed file as markdown does not support attaching python files.

fvrmr, 2021-09-09:
Hi @skabrits thank you for your report and solution. Did you also tested your solution?

skabrits, 2021-09-12:
Yes I have tested — it it works

Ghostkeeper, 2021-09-17:
Nice report. I'm a bit wary though that this change could cause other devices to also be detected as a printer, if they happen respond to something with `ok`. Or even `nokeyerror` or anything else that happens to have `o` and `k` next to each other. It would be safer to include more of the response in the check. Do you know what the complete response is of the printer?

And then to also fix the temperature read-out because if the message just contains `ok` then it won't parse a temperature for the monitor either.

Be aware that Ultimaker also doesn't maintain the USB printing workflow at all any more. We're merging fixes for it (like this one) but we have no way of testing it so we need to rely on the community to provide fixes for their printer models, like this.

closed 2024-12-09 by github-actions[bot]