Ultimaker/Cura #10412: Stl incompatible flashback

Cloud2702, 2021-09-08:
### Application Version

4.11.0

### Platform

windows 10

### Printer

custom fff printer

### Reproduction steps

Drag the stl model file into cura

### Actual results

Flicker occurs and cannot be sliced

### Expected results

Want stl model files to be compatible with all versions

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Voron V7测试方块.zip](https://github.com/Ultimaker/Cura/files/7130604/Voron.V7.zip)
official website can use this file to see if it will flash, and if it does, please improve it. The new version is not compatible with the stl model, so it cannot be sliced, otherwise you will have to continue slicing with a lower version.

fvrmr, 2021-09-09:
Hi @Cloud2702 thank you for your report. I can't reproduce your issue:

[image]

Could you share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Cloud2702, 2021-09-10:
2021-09-10 09:50:43,607 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-09-10 09:50:43,637 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdateChecker
2021-09-10 09:50:43,658 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-09-10 09:50:43,663 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdater
2021-09-10 09:50:43,663 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.11.0
2021-09-10 09:50:43,667 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2021 Ultimaker
2021-09-10 09:50:43,670 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-09-10 09:50:43,674 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2021-09-10 09:50:43,677 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2021-09-10 09:50:43,679 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2021-09-10 09:50:43,683 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2021-09-10 09:50:43,685 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2021-09-10 09:50:43,687 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-09-10 09:50:43,691 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2021-09-10 09:50:43,691 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2021-09-10 09:50:43,696 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2021-09-10 09:50:43,699 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2021-09-10 09:50:43,702 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2021-09-10 09:50:43,706 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-09-10 09:50:43,708 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2021-09-10 09:50:43,711 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2021-09-10 09:50:43,717 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzReader
2021-09-10 09:50:43,734 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzWriter
2021-09-10 09:50:43,751 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeProfileReader
2021-09-10 09:50:43,776 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeReader
2021-09-10 09:50:43,793 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeWriter
2021-09-10 09:50:43,811 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ImageReader
2021-09-10 09:50:43,826 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LegacyProfileReader
2021-09-10 09:50:43,841 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalContainerProvider
2021-09-10 09:50:43,858 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalFileOutputDevice
2021-09-10 09:50:43,873 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MachineSettingsAction
2021-09-10 09:50:43,889 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ModelChecker
2021-09-10 09:50:43,908 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MonitorStage
2021-09-10 09:50:43,926 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PerObjectSettingsTool
2021-09-10 09:50:43,941 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PostProcessingPlugin
2021-09-10 09:50:43,957 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PrepareStage
2021-09-10 09:50:43,973 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PreviewStage
2021-09-10 09:50:43,994 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RemovableDriveOutputDevice
2021-09-10 09:50:44,007 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SentryLogger
2021-09-10 09:50:44,036 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SimulationView
2021-09-10 09:50:44,054 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SliceInfoPlugin
2021-09-10 09:50:44,073 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SolidView
2021-09-10 09:50:44,092 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SupportEraser
2021-09-10 09:50:44,140 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin Toolbox
2021-09-10 09:50:44,155 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CameraTool
2021-09-10 09:50:44,174 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MirrorTool
2021-09-10 09:50:44,198 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RotateTool
2021-09-10 09:50:44,219 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ScaleTool
[... cut ...]
2021-09-10 09:54:09,644 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]:   File "messagestream.pyx", line 36, in scipy._lib.messagestream.MessageStream.__init__
2021-09-10 09:54:09,647 - ERROR - [JobQueueWorker [1]] UM.Logger.logException [116]: OSError: Failed to open file b'C:\\Users\\\xe5\xa4\xa9\xe7\xa9\xba\xe5\xb0\x8f~1\\AppData\\Local\\Temp\\scipy-_sf0a1de'
2021-09-10 09:54:09,652 - WARNING - [JobQueueWorker [1]] UM.Mesh.MeshFileHandler.readerRead [62]: Unable to read file C:/Users/天空小白云/Desktop/Voron V7测试方块.stl
2021-09-10 09:54:09,655 - DEBUG - [JobQueueWorker [1]] UM.FileHandler.ReadFileJob.run [83]: Loading file took 8.3 seconds

Cloud2702, 2021-09-10:
There is a lot of code in the log file. I'd better send the file. Please have a look at it.
[cura.log](https://github.com/Ultimaker/Cura/files/7140736/cura.log)

fvrmr, 2021-09-10:
@Cloud2702 thank you for your log file. 
Your issue is a duplicate of https://github.com/Ultimaker/Cura/issues/9644 and it is caused because the temp folder contains non-Latin characters which is located in the user folder, so it happens when the Windows username contains non-Latin characters.

I will close this issue as a duplicate.

closed 2021-09-10 by fvrmr

Cloud2702, 2021-09-11:
Do you mean that the user name must be in non-Chinese or English?

Cloud2702, 2021-09-11:
Did it your way, but still flinched.