Ultimaker/Cura #10413: Cannot change theme Cura 4.11

adalta87, 2021-09-08:
### Application Version

4.11

### Platform

Windows 10

### Printer

CR6-SE

### Reproduction steps

Tried changing the theme to dark and blind dark modes and restarted on both but the theme will not change in cura. only have added 4 printers cr6-se, snapmaker a350, cr10 and a tevo tornado. I also added 3 custom print profiles for the cr6-se

### Actual results

nothing

### Expected results

theme change but did not actually change

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NA

fvrmr, 2021-09-09:
Hi @adalta87 could your share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Ghostkeeper, 2021-09-17:
Still works okay as far as I can tell.

[image]

adalta87, 2021-09-17:
Sorry I have been away for awhile. Mine is now working as well odd that it didn't take after a cpl restarts but oh well its all good sorry for the delay!

[image]<https://user-images.githubusercontent.com/2448634/133777012-1f2a3f15-c1f9-4d95-9461-4ede9c0a0ba8.png>

Ghostkeeper, 2021-09-24:
Okay! With a log file we could've maybe seen what went wrong. But if you can't reproduce it any more either, we'll have to chalk it up as a mystery.

closed 2021-09-24 by Ghostkeeper