Ultimaker/Cura #10415: Supports generated by not interfaced.

balinhansen, 2021-09-08:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender-3 V2

### Reproduction steps

Install and configure Cura 4.8-4.11 with default settings for Ender-3 V2. (Ender-3 w/ 220x220mm bed)
Open example file with thin overhangs (rocket fins): [https://www.thingiverse.com/thing:3937496]
Enable Build Supports
Slice

*note that support structures are generated underneath the rocket's fins but there are no interfacing layers connecting support structures to said fins. I noticed this problem two years ago with Cura versions earlier than 4.8. I was only able to print this object because the flat spots on the fins held well enough. Rockets I'm working on that sit on points or edges are unprintable.

### Actual results

Printer nozzle knocks the fins over because there is nothing to support them.

### Expected results

Supports should be generated and properly interfaced to the overhangs. I am unsure why the default settings don't accomplish this or which settings I could modify to correct the problem. Decreasing  "Support Overhang Angle" accomplished nothing. Increasing "Support Overhang Angle" generates neither support structures under the fins nor interfacing media between the two which is the expected behavior.

Screenshot:

[image]

Project File:
[CE3_Hornet_Release.zip](https://github.com/Ultimaker/Cura/files/7131227/CE3_Hornet_Release.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Screenshot showing sliced object not connecting to support structure.

GregValiant, 2021-09-09:
Your minimum support interface area is 3.0 and that's too much for those angled fins.  At .5 there will be support interface.  You should probably kick up the interface area to 60% as well.

I'd print it on a raft or even a large brim.  Those fins don't need support, just good plate adhesion with Z-hops enabled at .4 high.  The print head lifting and then dropping vertically will keep the fins from being knocked over.

balinhansen, 2021-09-09:
Granted, with this particular model and good bed adhesion the fins print
with only minor issues and I'd imagine the z-hop setting should help and
I'll look into the support interface area setting. Thanks.

Guessing the macro view of the model isn't part of the way it procedurally
slices. It doesn't view the fins as a single surface which can be supported
so no interface is generated but it knows they're there and need support
somewhere it just doesn't know where.

Implementing a way to treat angled or even curved/round surfaces that could
usefully generate the interface as though it was the same flat overhang
layer and distributing said interface appropriate seems beyond the scope
for the way I'd imagine a practical slicer being programmed. Not be
impossible but yikes ... easier to let the default settings produce no
interface which is unintuitive especially since the support structures
still generate.

Maybe a way to prevent the support structures from being generated where
they aren't being used would give the user a stark visualization of the
lack of an interface would be easier to implement and alert them to a
problem in a way they'd immediately google and find solutions for like the
aforementioned:

"It's not a bug it's a feature ... so I'm not just going crazy or doing
anything wrong. It's just a edge case outlier. Thanks, Google!"

Thanks again.

fvrmr, 2021-09-09:
Hi @balinhansen thank you for your report. What @GregValiant mentioned is correct. You could also take a look at the `minimum support X/Y distance`.
You could also change your support to `tree support`. 

Let me know if that works for you!

closed 2021-09-23 by no-response[bot]