Ultimaker/Cura #10416: Symlink check for plugins

thomaskilian, 2021-09-08:
### Is your feature request related to a problem?

I do have a symlink for my own plugin that refers to some folder on my disk. When a new Cura version is loaded it does not recognize the symlink and issues a plain copy of the folder. 

### Describe the solution you'd like

A new version should check the folder for being a symlink and create another one in plugins.

### Describe alternatives you've considered

Well, I do that manually once I found out why my plugin changes did not take effect.

### Affected users and/or printers

Any

konskarm, 2021-09-14:
Hi @thomaskilian. We understand that this is a bit annoying, but after we discussed, we decided that it is not going to be prioritized any time soon if we put it in our already long backlog. However, we would be happy to review and test a PR with a solution if you feel like contributing to our codebase.

thomaskilian, 2021-09-14:
Fair enough. I haven't looked into Cura's code itself, but I'll see what time will give :-)

Ghostkeeper, 2021-09-17:
I'd suggest starting with this function, which does the copying: https://github.com/Ultimaker/Uranium/blob/2db3496f941f008bee053b3f4275a76ef8550e5f/UM/Resources.py#L485

thomaskilian, 2021-09-17:
Thanks! I already cloned the Cura sources and had a short look. Without that pointer it would have been hard to find. Especially when it's located in a totally different repo :-)

closed 2023-07-13 by github-actions[bot]