Ultimaker/Cura #10417: Tree supports are printed mid-air

RobVilchis, 2021-09-08:
### Application Version

4.10

### Platform

Windows 10

### Printer

Creality CR-10

### Reproduction steps

1. Dropped stl on build plate
2. Rotated by 180° on the X axis, then scaled by 113.76% (it's very big)
3. Sliced with default tree supports, at .24 mm layer height

### Actual results

Some parts of the tree supports are sliced to be printed mid air. The print might come out well, but it will create some spaghetti. It's not a rendering bug, that's how it prints.
[image]
[image]
[image]

### Expected results

All parts should be supported properly.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

WeTransfer link of project file:
https://we.tl/t-lF1pemV6Ik

fvrmr, 2021-09-09:
Hi @RobVilchis thank you for your report. 
This is a duplicate of https://github.com/Ultimaker/Cura/issues/8218 
Since it is a duplicate I will close this issue.

closed 2021-09-09 by fvrmr

RobVilchis, 2021-09-09:
Hello @fvrmr, thanks for your answer. I saw the issue you mentioned, but this was on version 4.10, not 4.7. And the unsupported parts seem much bigger and completely in the air, instead of being next to a wall. Maybe it's part of the same issue, but could you reconsider it? Thank you

fvrmr, 2021-09-09:
@RobVilchis the issue started in 4.7 and is not fixed yet in newer versions. We don't want to create for every version a new ticket, so it is linked to the one issue that was reported first. The ticket linked on our backlog is still in to do.