Ultimaker/Cura #10418: Crashing while slices

rockynewsmaster, 2021-09-08:
### Application Version

4.11.0

### Platform

Windows 10 Pro

### Printer

Ender 3 Pro

### Reproduction steps

Tried slicing a large object that sliced fine in 4.10

### Actual results

crashes and says Report a bug

### Expected results

it should slice and let me use the gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_Mason tube-morebevel4-rightthreads.zip](https://github.com/Ultimaker/Cura/files/7131495/CE3PRO_Mason.tube-morebevel4-rightthreads.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7131496/cura.log)

rockynewsmaster, 2021-09-08:
Created a new Low Profile and manually selected all the same (or so i think) settings and now i sliced. Uploading project for that as it shows the profile settings as well
[WORKING-CE3PRO_Mason tube-morebevel4-rightthreads-working.zip](https://github.com/Ultimaker/Cura/files/7131606/WORKING-CE3PRO_Mason.tube-morebevel4-rightthreads-working.zip)

smartavionics, 2021-09-08:
So, yes, it blows up the engine. I have found any one of the following actions allows the model to slice to completion...

1 - sink the model 1mm into the build plate.

2 - turn the model upside down.

3 - disable the Monotonic top/bottom order setting.

4 - use lines rather than zigzag skin pattern.

Hope this helps!

fvrmr, 2021-09-10:
I have created a ticket on our backlog for this. 
Developers see CURA-8553

closed 2021-10-12 by jellespijker