Ultimaker/Cura #10419: I can't scroll in, Add printer, Add a non-networked printer with the mouse wheel

Kitcarhans, 2021-09-09:
### Application Version

4.9 - 4.10 - 4.11

### Platform

Mac 10.15.7

### Printer

All

### Reproduction steps

On my Imac, 3.5GHz Quad-Core Intel Core i7, I can't scroll in, Add printer, Add a non-networked printer with the mouse wheel

### Actual results

-

### Expected results

-

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

-

fvrmr, 2021-09-09:
Hi @Kitcarhans does this also happen with the touchpad or only with external mouse?

Kitcarhans, 2021-09-09:
When using a mouse. I don't know about a touchpad.

Met vriendelijke groet,
Hans de Bruijn
06028107780

Verzonden vanaf mijn iMac.

fvrmr, 2021-09-09:
Can you share your log file? You can find your log file here: `$HOME/Library/Application Support/cura/<Cura version>/cura.log`

Kitcarhans, 2021-09-09:
Here the log file

Met vriendelijke groet,
Hans de Bruijn
06028107780

Verzonden vanaf mijn iMac.

fvrmr, 2021-09-10:
I can't see your log file, since you've replied via e-mail. Please post the logs in a comment here on Github.

Kitcarhans, 2021-09-10:
[cura.log](https://github.com/Ultimaker/Cura/files/7142841/cura.log)

fvrmr, 2021-09-10:
What kind of mouse are you using? The Apple Magic Mouse? And did this happen in previous versions as well?

Kitcarhans, 2021-09-10:
I'm using the MX Master 3 mouse from Logitech and it happens from 4.8 for the Mac. I haven't used versions before that on a Mac.

Kitcarhans, 2021-09-10:
I use a Logitech mouse MX Master 3. From 4.8

Met vriendelijke groet,
Hans de Bruijn
06028107780

Verzonden vanaf mijn iMac.

Ghostkeeper, 2021-09-17:
One possible cause might be that there are actually two overlapping scroll areas there, as can be seen by the little one behind the main scroll bar:
[image]

However as far as I know, this only started with the Qt upgrade, which would be in 4.9. Since the problem apparently also happens in 4.8, maybe that's not what is causing this.

We also have a long-standing issue where it's just hard to scroll. Not impossible, but the scrolling is very slow because somehow it takes the height of the hidden printers into account as well. But that's not what Hans is reporting (although we may be misinterpreting this).

The log file is silent about this. There are no QML errors in the dialogue in question. So at least Qt must be happy with it. There may be something else in the QML that we designed wrongly of course.

I'll ask the team what they think.

Ghostkeeper, 2021-09-20:
We've tried to reproduce this, on two Macs, one of which happened to have the exact same mouse model. In both cases it was scrolling very slowly, but it was still scrolling. So far we haven't been able to reproduce it then. We'll have to wait and see if anyone else can also reproduce this issue, to see if there's a pattern to who has the issue and who doesn't.

closed 2024-12-09 by github-actions[bot]