Ultimaker/Cura #10420: Ultimaker printer said "Job was encoded by older CURA program, please use Cura4.4 or newer...  blablabla"

behpaja, 2021-09-09:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ultimaker S5 with Bot image sides

### Reproduction steps

1. create cut on Cura 4.11
2. Load job on the USB Key
3. activate job on printer
4. push Ignore on the printer panel

### Actual results

Since Cura 4.10 I have this window message
I push Ignore and I can print my job

### Expected results

No need to push Ignore with Cura 4.11

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Type: Ultimaker S5
firmware of the printer: 5.5.12-20200226-ultibot
Id Type material: 9051-0

fieldOfView, 2021-09-09:
This has been fixed in a newer version of the printer firmware. Please update your printer(s) to the latest firmware.

behpaja, 2021-09-09:
Hello,
Have you got an link for the firmware?

Thanks
Pascal.

Pascal JANSSEN
RST Coordinator
Wet wipes
ESSITY BELGIUM SA-NV
Phone +32 87 306 729
Mobile +32 496 127 603
***@***.******@***.***>

Essity Internal

This message may contain confidential, proprietary or legally
privileged information and is intended for the addressee's use
only. Any usage, disclosure, distribution, print or copying of
any part of this message is prohibited unless you are the
intended recipient. If you receive this message in error,
please delete it from any computer and notify the sender. If
you suspect that this message may have been altered, please
notify the sender. Essity has taken every reasonable precaution
to ensure that this e-mail and any attachments to this e-mail
has been scanned for viruses. However, Essity does not accept
liability for any damage caused by software viruses brought to
you by this mail.

behpaja, 2021-09-09:
Hello,
I found them
Thanks
Pascal.

Pascal JANSSEN
RST Coordinator
Wet wipes
ESSITY BELGIUM SA-NV
Phone +32 87 306 729
Mobile +32 496 127 603
***@***.******@***.***>

Essity Internal

fvrmr, 2021-09-09:
Hi @behpaja  I will close this issue since it is solved with a firmware update. 
Thank you @fieldOfView

closed 2021-09-09 by fvrmr