Ultimaker/Cura #10421: [4.11.0] Material Display Name Isn't Used Anymore For Top Bar

mzhu1113, 2021-09-09:
### Application Version

4.11.0

### Platform

Win 10

### Printer

N/A

### Reproduction steps

N/A

### Actual results

N/A

### Expected results

Expected previous behavior where the Material Display Name was used*

*A little more clarification:
We like that 4.11.0 is actually using the Brand/Material from the material properties now (We had to workaround that by putting that same info in the Material Display Name property in prior versions).

However we utilize slightly different temperatures based on layer height used (and at times part complexity) for our slicers. Hence in our screenshot we have the layer height in the Material Display Name property. We'd like the Material Display Name to come back, it doesn't have to be the same behavior like in prior versions where it's only that property but a compromise/hybrid would be fine so long as we can get back that property for the top bar.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-09-09:
Hi @mzhu1113 can you share your material profile with me?
So I can have a look on it.

mzhu1113, 2021-09-10:
[ABS Generic ZLabs.xml.zip](https://github.com/Ultimaker/Cura/files/7140369/ABS.Generic.ZLabs.xml.zip)
Here you go.

fvrmr, 2021-09-10:
Thanks, I will bring this up with our UX designer.

lorenzoromagnoli, 2021-09-13:
Hi @mzhu1113, we already have a ticket on the backlog (CURA-8565) to fix this behaviour.  We want to show the same name in the top bar and in the material selection dropdown 

if label is set we will show <brand>+<material label (aka display name)> else it is going to be <brand><color><type> as it is now.

mzhu1113, 2021-09-14:
@lorenzoromagnoli Alright sounds good. Sounds like no changes needed from my end. Appreciate it.

commit 2baada8a (2021-10-20): Use the material name instead of brand + type + color

CURA-8514
Fixes #10421
closed 2021-10-20 by Ghostkeeper