Ultimaker/Cura #10422: No correct Tempsetting for Initial Layer at Nozzle an Build Plate

horstwill, 2021-09-09:
### Application Version

4.10.0

### Platform

Windows10 and Windows 7 (both tested)

### Printer

Creality CR-X

### Reproduction steps

Set Printing Temperature to 230. Set Printing Temperature Initial Layer to 210. Set Initial Printing Temperature 210.
[image]

### Actual results

The Gcode only ever reflects a temperature of 230. 210 never appears in Gcode.

### Expected results

There should be a "M104 S210.0" in Gcode before Layer 0. Later on there should be a "M104 S230.0"

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Nadelfeilen 12pcs.zip](https://github.com/Ultimaker/Cura/files/7138049/Nadelfeilen.12pcs.zip)

GregValiant, 2021-09-09:
In both your regular StartUp Gcode and the extruder change StartUp Gcode you specifically call out {material_print_temperature}.  In the StartUp Gcode that should be {material_print_temperature_layer_0}.
I think the M104 line needs to be removed from the Extruder Change startup Gcode.
Your StartUp Gcode also has a T0 line.  I'd remove that.
That is a 2 in 1 out printer?  In Printer Settings I think you need to enable "Extruders Share Heater" and "Extruders share Nozzle" as well.

horstwill, 2021-09-12:
Thanks for the help.
Removing the M104 command has the effect that suitable temperature commands are written into the G-code.

closed 2021-09-12 by horstwill