Ultimaker/Cura #10423: Invalid printing time

Trophy-924, 2021-09-09:
### Application Version

4,10 ; 4,11

### Platform

Windows 7

### Printer

Anet A8 Plus

### Reproduction steps

Slicing, printing.
[CFFFP_Bitholder 20mm d.s  .zip](https://github.com/Ultimaker/Cura/files/7138349/CFFFP_Bitholder.20mm.d.s.zip)

### Actual results

Invalid printing time.

### Expected results

The indicated and actual time should be the same. (more or less)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

After slicing, the displayed print time in Cura does not match the actual print time in versions 4,10, and 4,11.
Only these two versions had this kind of problem, there was no problem with the previous versions I used. 
Before writing the bug report, I printed the part visible on the screenshot.
According to Cura 4,11,0, the estimated printing time is 5 hours 57 minutes. 
The actual printing time was 7 hours 23 minutes.
[image]

[image]

It doesn't matter what I print! Estimated and actual print times never approach each other !
I also tried printing from a memory card and USB. The end result is the same.

fvrmr, 2021-09-10:
Hi @Trophy-924 thank you for your report. The printing time is calculated on the printer settings in Cura. We don't and can't test third party printers, so unfortunately we can't really help you with this. 
That's why I will close this issue. I hope you understand.

closed 2021-09-10 by fvrmr

Ghostkeeper, 2021-09-17:
Try letting Cura take control over the acceleration and jerk. If Cura doesn't have control, it'll use the firmware defaults. Those firmware defaults are not always included in the profiles users submitted for their printers, so Cura doesn't always have the correct defaults.