Ultimaker/Cura #10424: Cutting mesh modifier changes wall settings for unknown reasons

BloodyRain2k, 2021-09-09:
### Application Version

4.11

### Platform

Win 7 x64

### Printer

CR-6

### Reproduction steps

I was trying to only turn off the ironing setting for a certain height of the model, so I added a support blocker to have a mesh for override modifiers.

But when I changed it from "Infill Only" to "Cutting" I immediately got the weird result below.

### Actual results

[image]

### Expected results

Walls without gaps.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[SlideCutter.zip](https://github.com/Ultimaker/Cura/files/7138435/SlideCutter.zip)

BloodyRain2k, 2021-09-09:
I've just checked this back in 4.9.1 and there the problem also exists, and it looks like the walls inside the overlapping mesh get printed thinner, for whatever reason...

fvrmr, 2021-09-10:
Hi @BloodyRain2k thank you for your report.
This is expected behaviour with a cutting mesh. You can change the wall thickness for the cutting mesh but than you still have walls on the other model. I would see this as a feature request then, I will bring it up with the team. Keep you posted!

BloodyRain2k, 2021-09-10:
I fail to see this as expected behaviour when I have NOT chosen to change the wallthickness...
Worse: even IF I choose to do so, the values that create the walls outside the mesh create a different result inside the cutting mesh...

Ghostkeeper, 2021-09-14:
This is indeed not expected behaviour. I think they misunderstood what your actual issue was.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9083.

As a workaround you can disable the Enable Bridge Settings setting for this print.

closed 2021-09-14 by Ghostkeeper