Ultimaker/Cura #10425: Cura 4.11.0 First layer test STL shows 0 minutes to print but in older cura is show the right time to print

alcomys, 2021-09-09:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Anet A8 PLUS

### Reproduction steps

Load an test STL file and the new cura show 0 minutes to print and not showing an preview
In the version before 4.11.0 it works well and show me 8 minutes to print time.

### Actual results

Nothing it wont print because it shows zero minutes to print.

### Expected results

The result must be that i got the right printing time and not show zero time

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-09-10:
Hi @alcomys it looks like this is a duplicate of https://github.com/Ultimaker/Cura/issues/10410 
I can't see the height of your model from the screenshot? If your object is above 0.3 mm it should works, from looking at your profile.

alcomys, 2021-09-10:
Hi fvrmr,
you are right but the solution is ridiculous to increase the layer height
this problem resist after the 4.11.0 update so if you use the 4.10.0 the problem is gone.
So i wait till there is a better solution or an upgrade of cura

hitboxr, 2021-09-11:
It's showing 0 minutes print time because the model is shorter than the layer height, this making the model un-printable with the current configuration. You're trying to use a 0.3mm layer height profile (top right of your screenshot) with a model that is 0.2mm tall (bottom left). This is not a bug, it's Cura functioning as intended. Every recent version of Cura behaves exactly in this way. This should probably be closed.

hitboxr, 2021-09-11:
This happens in 4.10 as well actually, so doesn't seem to have been introduced in 4.11

smartavionics, 2021-09-11:
Here's a fix for this bug but I wonder if this logic needs to take into account the slicing tolerance setting. i..e if the tolerance is set to middle and initial_layer_height/2 <= model_height <= initial_layer_height, should it print the first layer or not?

```
diff --git a/src/FffPolygonGenerator.cpp b/src/FffPolygonGenerator.cpp
index 1075e07e..e2fb5191 100644
--- a/src/FffPolygonGenerator.cpp
+++ b/src/FffPolygonGenerator.cpp
@@ -132,7 +132,7 @@ bool FffPolygonGenerator::sliceModel(MeshGroup* meshgroup, TimeKeeper& timeKeepe
     else
     {
         const coord_t height_without_first_layer = storage.model_max.z - initial_layer_thickness;
-        if(height_without_first_layer <= 0)
+        if(height_without_first_layer < 0)
         {
             slice_layer_count = 0;
         }
```

alcomys, 2021-09-11:
Hello john.

 

It does not happen to me in cura 4.10.0

Only in 4.11.0

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

alcomys, 2021-09-11:
I did try to use the normal 0.2 layer height in cura 4.11.0 and still not slice.
I try the same in cura 4.10.0 with the layer height 0.2 and the cura will slice it.
so it has not to do with the layer height

theFrantic, 2021-09-13:
I can confirm this problem on Cura 4.11, I've downgraded to 4.10 in order to avoid it. If you tray to print any bed level stl (that are a one layer only print normally) it happens.

I'm providing two gcode files (one for the 4.11 and other for the 4.10 version) if it helps to detect the problem. As far as I've tested this only happens with one layer prints, it slices bigger prints.

[gcode_files.zip](https://github.com/Ultimaker/Cura/files/7154428/gcode_files.zip)

BenDavidson90, 2021-09-14:
Same problem here, have to revert to 4.10 in order to continue my test prints.

pulponair, 2021-09-14:
Confirmed. Super annyoing.

Ghostkeeper, 2021-09-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/10394.

I made an analysis of this bug and the expected behaviour here: https://old.reddit.com/r/Cura/comments/povb83/cura_411_does_not_slice_bed_leveling_files_anymore/hd3h73n/

closed 2021-09-17 by Ghostkeeper