Ultimaker/Cura #10426: Crashed GUI after new update

azorek777, 2021-09-09:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

I installed a new version 4.11 then uninstalled the previous 4.10. 

### Actual results

The first run was fine, everything worked fine. After launching the computer a day later Cura GUI crashed and looks like this
[image]
Changing the theme doesn't help (for example Ultimaker Dark)
[image]

### Expected results

Cura should works like at first launch (it means fine)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7139483/cura.log)

fvrmr, 2021-09-10:
Hi @azorek777 do you have problems with other programs that are using OpenGL?

azorek777, 2021-09-15:
No, I uninstalled and installed it again. Then it started to work

Ghostkeeper, 2021-09-17:
Maybe the installation failed then for some reason, the first time? It didn't install all of the files or something? Or maybe it was installed on top of a previous installation?

closed 2021-09-17 by Ghostkeeper