Ultimaker/Cura #10428: Cura 4.11 Close when open new file

gusdudu, 2021-09-10:
### Application Version

4.11

### Platform

windows 10

### Printer

Artillery X1

### Reproduction steps

Cura 4.11 Close when open new file, but drag and drop is OK.
the error ocurs when you activate **PREVIEW MODE IN WINDOWS**, because if you desactivate **PREVIEW PANNEL** in windows, files open OK.

### Actual results

app close

### Expected results

app close

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

the error ocurs when you activate PREVIEW MODE IN WINDOWS, because if you desactivate PREVIEW PANNEL in windows, files open OK.

fvrmr, 2021-09-10:
Hi @gusdudu could you share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

userosos, 2021-09-10:
I confirm it. Windows 10 64 bit.
[cura.log](https://github.com/Ultimaker/Cura/files/7144309/cura.log)

stephanp66, 2021-09-11:
Hi, I have the same trouble.
Win10 64bit , Artillery X1
[cura.log](https://github.com/Ultimaker/Cura/files/7148158/cura.log)

nallath, 2021-09-12:
This isn't an issue caused by us but by Chitubox. We can't actually fix this, but we have written a help page explaining how you can work around this issue (https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file)

Please raise this issue with Chitubox for a full fix.

closed 2021-09-12 by nallath

jellespijker, 2021-09-12:
Duplicate of #9668