Ultimaker/Cura #10430: 3mf file support enforcer printed

peteakalad, 2021-09-10:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I was attempting to print the files for the prusa v2 enclosure, opening .3mf files in prusa.

For some reason the support enforcers which are visible as such in prusa slicer, are treated as printable in cura.  Unfortunately I'd not noticed as I didn't know the models and printed them.  Only after when I opened in prusa slicer did I notice the problem.

To reproduce open prusa_mmu_enclosure_1.3mf, for example the small hinges at the front of the bed, should not have cubes around them, in prusa slicer these are seen as support enforcers.
https://media.prusaprinters.org/media/prints/3673/stls/158821_5e268306-21c6-4a44-acdb-117fd7946322/prusa_mmu_enclosure_1.3mf

(Many thanks for Cura, great s/w!)

### Actual results

Cura treats 3mf support enforcers as printable.

### Expected results

Cura treats 3mf support enforcers as such, or exclude them so they are not erroneously printed.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

https://media.prusaprinters.org/media/prints/3673/stls/158821_5e268306-21c6-4a44-acdb-117fd7946322/prusa_mmu_enclosure_1.3mf

GregValiant, 2021-09-12:
Select the part.  Right click and select Mesh Tools (you must have the plugin loaded).  Select "Split Model into Parts".
Select the portion that is supposed to be a blocker and go to Per Model Settings and select "Don't Support Overlaps".

It appears that PrusaSlicer or Cura (or both of them) aren't in tune with whatever standards apply to describing blockers in a 3mf file.

peteakalad, 2021-09-12:
Ok thanks for the workaround. Had been using Prusa slicer to remove the support enforcers, saving and opening in Cura to slice, but if I can do similar all in Cura that’s better.  Although may still need Prusa to determine which parts are those support enforcers. 

Yes hopefully these non printed 3mf category object parts can be excluded/ignored by Cura as a simple fix.

nallath, 2021-09-12:
Sorry; Prusa does this in such a way that we can't fix this. Instead of adding support blockers as separate meshes, it adds them inside a given mesh (which, as far as I understood it, isn't allowed according to the 3mf spec. Even if it is, it does mean that their way of working will give issues with all other 3mf implementations). I've ran into the exact same issue when I was printing this (and that's also how I know how they handle modifier meshes).
 
They use a prusa specific extension to 3mf to read this back again, making it incompatible with everything else. 

So this is something you will have to take up with them I'm afraid. Feel free to tag me in the ticket if you decide to raise it. Id be happy to discuss the issue with their developers if need be.

closed 2021-09-12 by nallath