Ultimaker/Cura #10433: Support Blockers are not selectable when reloading a Cura project file.

DLLarson, 2021-09-11:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Creality Ender-3 V2

### Reproduction steps

I created several support blockers and saved the file as a Cura project file. When I reloaded the file I could see that the support blockers show up as part of the model but can't be selected:

[image]

-Dale

### Actual results

When I run the slice operation the areas with the blocks are preventing supports but I can't select or adjust them at all.

### Expected results

I expect that I can modify, add  or delete support blockers between Cura project file load sessions.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file for bug.

[CE3_swoopingnight.zip](https://github.com/Ultimaker/Cura/files/7148358/CE3_swoopingnight.zip)

-Dale

GregValiant, 2021-09-12:
Either ungroup the model and select the blocker or you can use CNTRL+Lt Click to select just the blocker in the group.

DLLarson, 2021-09-13:
If I ungroup the model the support blockers lose their position and fall onto the bed. However the Ctrl-Lbutton click works so that will enable me to proceed. I do get lost in the UI at times. Weirdly. I can't seem to locate a comprehensive manual for such a popular program.

When I try to add an additional blocker which should be 1cm x 1cm it creates a cube that is over 15cm square-ish and is not exactly cube shaped:

[image]

The model was scaled up 1500% right after import. The rescaling didn't give me issues when I added prior support blocker blocks. Something seem out of whack.

nallath, 2021-09-13:
I think this might be related to https://github.com/Ultimaker/Cura/issues/9973

DLLarson, 2021-09-13:
>I think this might be related to #9973

Yep. That's exactly what happens. I'm running on Windows 10 although someone in that issue said it was working on Win10 for him.

closed 2021-09-13 by nallath