Ultimaker/Cura #10434: Cura 4.11 not running with pyqt5 15

berberic2, 2021-09-11:
### Application Version

4.11

### Platform

Linux

### Printer

None

### Reproduction steps

install/compile cura (cura 4.11, uranium 4.11, libarcus 4.11, cura-engine 4.11) on a system with pyqt5 5.15.2.

start cura

### Actual results

> cura
/usr/lib/python3/dist-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 57, in <module>
    from cura.Arranging.ArrangeObjectsJob import ArrangeObjectsJob
  File "/usr/lib/python3/dist-packages/cura/Arranging/ArrangeObjectsJob.py", line 11, in <module>
    from cura.Arranging.Nest2DArrange import arrange
  File "/usr/lib/python3/dist-packages/cura/Arranging/Nest2DArrange.py", line 5, in <module>
    from pynest2d import Point, Box, Item, NfpConfig, nest
ValueError: PyCapsule_GetPointer called with incorrect name

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/cura", line 187, in <module>
    from cura.CuraApplication import CuraApplication
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 57, in <module>
    from cura.Arranging.ArrangeObjectsJob import ArrangeObjectsJob
  File "/usr/lib/python3/dist-packages/cura/Arranging/ArrangeObjectsJob.py", line 11, in <module>
    from cura.Arranging.Nest2DArrange import arrange
  File "/usr/lib/python3/dist-packages/cura/Arranging/Nest2DArrange.py", line 5, in <module>
    from pynest2d import Point, Box, Item, NfpConfig, nest
ValueError: PyCapsule_GetPointer called with incorrect name

### Expected results

running instance of cura

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

smartavionics, 2021-09-11:
FYI, Cura releases are currently being built with PyQt 5.15.2 so that version of PyQt is compatible.

berberic2, 2021-09-14:
My fault.

closed 2021-09-14 by berberic2

Ghostkeeper, 2021-09-20:
If you're willing to share what you did to fix the issue, that might help future contributors trying to deal with this issue if they make the same mistake.

berberic2, 2021-09-20:
Sure: My version of pynest2d was too old.
I upgraded all packages to 4.11 but forgot pynest2d.

Ghostkeeper, 2021-09-21:
Ah, yes indeed. Then I suspect https://github.com/Ultimaker/pynest2d/pull/9 was the fix for the issue.