Ultimaker/Cura #10435: Cura stable crash while slicing stl with 0.2mm layer height with own slice settings

Mistpal, 2021-09-11:
### Application Version

4.11

### Platform

Windows 10

### Printer

Flying Bear Ghost 5

### Reproduction steps

1. Import my cura profile 
2. add file base_bottom_layers.STL into Cura
2021-09-12 02:28:21,154 - DEBUG - [JobQueueWorker [7]] UM.Logger.logException [116]: NameError: name 'glass' is not defined

### Actual results

Cura slice process is crashed

### Expected results

Cura sliced the model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Cura slice process stable crashing and after restart Cura
i just tried to increase the model by Z from 0.2mm to 0.25 - Cura NOT CRASHING

[cura_log_profils.zip](https://github.com/Ultimaker/Cura/files/7148490/cura_log_profils.zip)

GregValiant, 2021-09-12:
This is getting to be a popular problem with calibration shapes that are the same exact thickness as the layer height.
With Uniform Scaling turned off make the Z .21 and it will slice.  On Monday someone from the team will take a look.

Ghostkeeper, 2021-09-20:
Well, Greg didn't specify which Monday...

It looks like this is actually a special case of https://github.com/Ultimaker/Cura/issues/10394 due to an off-by-one error where it'll try to access a layer number that doesn't exist. It should be fixed at the same time.

closed 2021-09-20 by Ghostkeeper