Ultimaker/Cura #10436: Crashes on loading any stl or gcode file

duindain, 2021-09-12:
### Application Version

Every version after 4.6.2

### Platform

Linux Mint 20.2

### Printer

Creality CR10

### Reproduction steps

Download app image file
Right click on file and set to Allow executing as a program
Run the AppImage
Drag and drop any stl or gcode file onto the cura window

### Actual results

A fatal error has occured in Cura crash report popup loads
Clicking Send Report or Close closes Cura

### Expected results

Cura should load the stl or gcode file and allow normal operations

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I've been using Cura 4.6.2 since it was released, all versions after this demonstrate identical behaviour crashing as soon as a file is opened
I've always submitted the bug reports within the program hoping to get this resolved

[cura.log](https://github.com/Ultimaker/Cura/files/7148914/cura.log)

duindain, 2021-09-12:
[image]

nallath, 2021-09-12:
We've had a corruption with some data a while ago. It could be that that you still have that issue (so corrupted data is loaded in). That would also explain why we don't get the crash reports; We've marked that as to be ignored, since it has been fixed.

There are two options;
1. Remove your configuration folder (all of it) and try again. The downside obviously being that this means you lose all your data.
2. Upload your configuration folder here. We should be able to fix the issue by hand

duindain, 2021-09-13:
Hi @nallath, thx for the help

I've renamed the ~/.local/share/cura folder and can now open the latest version of cura and open stl and gcode files

Thanks

closed 2021-09-13 by duindain