Ultimaker/Cura #10438: Infill

AndyCap777, 2021-09-12:
### Application Version

4.11

### Platform

Windows 10

### Printer

Creality Ender 3v2

### Reproduction steps

1. I loaded my model into Cura 4.11 
2. I hit preview and rolled through the layers

### Actual results

I don't want to print because of the fear of it coming out like it looks in the slicer
[image]

### Expected results

Like in version 4.10
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

AndyCap777, 2021-09-12:
Infill seems to be corrupt

closed 2021-09-12 by AndyCap777

Ghostkeeper, 2021-09-20:
Maybe you had the Infill Support setting enabled?