Ultimaker/Cura #10440: Failed to slice.

nffxboy, 2021-09-12:
### Application Version

4.11.0

### Platform

Win 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

Uploaded my design used attached profile and tried to slice, during slicing process error appears.

### Actual results

Failed to slice

### Expected results

Slicing completed

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[output_log.txt](https://github.com/Ultimaker/Cura/files/7150373/output_log.txt)
[image]
[1-Main.zip](https://github.com/Ultimaker/Cura/files/7150375/1-Main.zip)
[.cura.zip](https://github.com/Ultimaker/Cura/files/7150376/default.cura.zip)
[Profile.zip](https://github.com/Ultimaker/Cura/files/7150377/Profile.zip)

Ghostkeeper, 2021-09-14:
I'm unable to reproduce this problem. It slices fine here, using your profile and STL, rotated 45 degrees to fit on the build plate.

[image]

However a profile and STL are not enough to completely reproduce the issue. What I'd also need to know is:
* You are using a custom material profile. In the screenshot it's visible that you're using eSun PLA+, but none of the colours of eSun PLA+ available by default are compatible with the profile you're giving, so you must have duplicated one of those profiles and edited the Material Type or something?
* The precise position and orientation of the model.
* The nozzle profile you're using.

Luckily, we've made a handy button to save all of this at once. Could you go to File -> Save Project and upload the resulting file here? It contains all of the profiles you're using, all settings you modified on top of that, all models and the precise position and orientation of them. This way we can reproduce your slice exactly.

closed 2021-09-28 by no-response[bot]

nffxboy, 2021-10-16:
[CE3PRO_Helmet horns L.zip](https://github.com/Ultimaker/Cura/files/7358264/CE3PRO_Helmet.horns.L.zip)

attached