Ultimaker/Cura #10441: Cura 4.11.0 Layer Shifting

cjmalchow, 2021-09-13:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

I've experienced multiple times where this new upgrade to Cura has resulted in layer shifting of the prints. It seems more likely to happen the more full the build plate is. I will be reverting back to 4.10.0 and will update this bug report with whether it happens after downgrading. Quite simply all I do is set the settings I want, slice it, upload to OctoPrint and start printing.

### Actual results

Layer shifting occurs intermittently. The shifting is not tiny, it shifts approximately 1cm.

### Expected results

It should print the object without any layer shifting, especially to the extreme I have been experiencing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

I have tried to upload the project (3MF file) but this system says it's not a supported file type. Please advice how you would like to receive the project file. Thanks

GregValiant, 2021-09-13:
In Cura clear the build plate and open the gcode file.  Cura doesn't remember how it created the gcode file and the gcode reader is separate anyway.  You will get a faithful preview of the gcode (so long as you didn't use the ArcWelder plugin).  If it shows the layer shifts then zip the 3mf file and the gcode file and post it here.  I can read gcode files into AutoCad even if they were created using ArcWelder and that takes Cura out of the loop.

99.9% of the time the cause layers shifting is loose belts but 99.9% of the time it isn't 10mm all at once.

cjmalchow, 2021-09-14:
Yeah, I checked my belts and it's also intermittent. So, I'd be super
surprised if that's what it ended up being. I'll upload a copy of the GCode
and send it to you as I do use ArcWelder. I'll also attach the 3mf. I
should be able to upload shortly. Thanks!

cjmalchow, 2021-09-14:
[CE3PRO_White_2.zip](https://github.com/Ultimaker/Cura/files/7158620/CE3PRO_White_2.zip)
I hope these are the right files :-S

cjmalchow, 2021-09-14:
I did open the GCode in NCViewer.com and I can see the layer shift.

Ghostkeeper, 2021-09-14:
> [CE3PRO_White_2.zip](https://github.com/Ultimaker/Cura/files/7158620/CE3PRO_White_2.zip)
> I hope these are the right files :-S

They are. Except that I get very different g-code if I slice it because you're using Arcwelder.
From the slice I can't immediately find an issue with it. Looks normal to me. There is no instruction to shift the subsequent layers, in any case.

You are, though, asking us to help with a printing process issue. We don't have your printer so we can't reproduce the print. All we can really do from here is look at the g-code, but the g-code is a poor approximation of how it will actually print. To really have a bug report, we'd need to know what Cura is doing wrong.

The Settings Guide plug-in has an article about this issue and what you can do to prevent it, both in hardware and in Cura's settings: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/layer_shift.md
Other than that, there's really not a lot I can debug for you unless you can point out what in Cura's output is wrong.

closed 2021-09-14 by Ghostkeeper

Ghostkeeper, 2021-09-20:
I used the Cura preview and the RepetierHost preview in my tests to determine the result I said before. The g-code contains no explicit layer shift.

I'm fairly sure that the layer shift is just a physical layer shift caused by too high accelerations, too low motor power or the nozzle bumping into things like blobs on the print. There's not a lot we can debug then, unless Cjmalchow can point out what is wrong in the g-code, or give suggestions as to how the g-code should be different.

dukeduck1984, 2021-12-11:
Hi @cjmalchow , have you figured out where the issue was?  I'm having the same issue - massive layer shifting in large prints.  It happened to me 3 times.  The first time I tought it was the belts, and I replaced some new belts and carefully routed and tightened, and had a couple of smaller prints as tests which turned out flawlessly as usual, and then in an almost full plate 20+ hours large print, it just happened again.  Like you said: `It seems more likely to happen the more full the build plate is.`

My printer is voron 2.4 350mm, with the lastest klipper, cura 4.11

How's your test with cura 4.10?